realthunder/FreeCAD_assembly3 #204: Subshape binders are created at wrong place if source is a link

ceremcem, 2019-03-19:
# Reproduction 

1. Create a link of `allen bolt` in `main` project. 
2. Right click on `allen bolt` link, select "Link actions > Make link"
3. Place those two linked objects freely
4. Create a subshape binder for each one of them. 
5. See all subshape binders are created at the same point

[image]

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16482 (Git)
Build type: Release
Branch: LinkStage3
Hash: f670d46f2d1255867b3330052496894cf9ff7a23
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

[link-test-20190319T2325-wrong-subshape-binder.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2985456/link-test-20190319T2325-wrong-subshape-binder.zip)

realthunder, 2019-03-26:
Bug fixed. You need to manually recompute the binder to fix it. Thanks for reporting!

closed 2019-03-26 by realthunder

ceremcem, 2019-03-26:
It seems solved, thanks. 

However, this might be a good case to bring the "Test Suite" necessity forward. Do you think that will require a scripting feature like DeclaraCAD has?

realthunder, 2019-03-28:
FC already has a test workbench for this kind of thing that uses Python unittest. And most of the problem reported here are not really modeling problem.

On the other hand, I think cadquery is more suitable for scripting models in FC, which is fully parameterizable with the new expression enhancement.