realthunder/FreeCAD_assembly3 #205: Offset option for PointsOnCircle constraint

ceremcem, 2019-03-20:
It would be useful to have an optional offset option (zero by default, like PlaneAlignment constraint)  for bolt placement operations.

realthunder, 2019-03-26:
The `Offset` property in `PlaneAlignment` is obsolete. You can now have distance/angle offset on any constraint by moving the constraining element. If you change the `Offset` property of an element link under the constraint, then only that constraint is affected. If you change the `Offset` property of the linked element under the `Elements` group, than all constraints using that element will be affected.

ceremcem, 2019-03-27:
Got it: 

[image]

This approach adds a little bit complexity but it sounds like a more filexible solution. 

Thanks.

closed 2019-03-27 by ceremcem