realthunder/FreeCAD_assembly3 #206: Group behavior in containers

ceremcem, 2019-03-20:
In order to reduce the tree view complexity (make the long tree short), it seems like an obvious approach to use the "Group"s. However, when we move some items into a "Group", solver can no longer find those components: 

[image]

Is it possible to group the components while interpreting like there is no grouping is made (they are residing on the same level before grouping)?

ceremcem, 2019-03-24:
Duplicate of #134 (but this one adds more context)

realthunder, 2019-03-26:
Yes, I think that's a good use case for the plain `Group`. I'll add support for that soon.

realthunder, 2019-04-08:
I have just add support for plain group inside assembly. Please sync both the LinkStage3 and asm3 repo. And also please test it well, as it is not a trivial feature. Also, please note that although the original FC group now works in asm3, you should use the asm3's new toolbar button [image]. It has some extra feature, like sorting children, join children on delete, etc.

ceremcem, 2019-04-09:
# Screencast 

[image]

# Backtrace 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f7eabbb8060]
#1  0x7f7eb4c4c445 in App::ExtensionContainer::hasExtension(Base::Type, bool) const from /opt/FreeCAD/lib/libFreeCADApp.so+0x15
#2  0x7f7eb4d2e836 in App::LinkBaseExtension::flattenSubname(char const*) const from /opt/FreeCAD/lib/libFreeCADApp.so+0xa6
#3  0x7f7eb4d427b1 in App::LinkBaseExtensionPy::flattenSubname(_object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x41
#4  0x7f7eb4d42ad3 in App::LinkBaseExtensionPy::staticCallback_flattenSubname(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x23
#5  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541) [0x7f7eb4141091]
#6  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f7eb413f390]
#7  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#9  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#10  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e670) [0x7f7eb41fc670]
#11  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#12  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193dfc) [0x7f7eb4251dfc]
#13  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#14  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7eb42a76c7]
#15  0x7f7eb4cf58cb in App::FeaturePythonImp::execute() from /opt/FreeCAD/lib/libFreeCADApp.so+0x18b
#16  0x7f7e8b6dd085 in App::FeaturePythonT<Part::Feature>::execute() from /opt/FreeCAD/lib/Part.so+0x15
#17  0x7f7eb4c41e19 in App::DocumentObject::recompute() from /opt/FreeCAD/lib/libFreeCADApp.so+0x79
#18  0x7f7e8b6cf0db in Part::Feature::recompute() from /opt/FreeCAD/lib/Part.so+0xb
#19  0x7f7eb4be055d in App::Document::_recomputeFeature(App::DocumentObject*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x2ed
#20  0x7f7eb4bf8bac in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /opt/FreeCAD/lib/libFreeCADApp.so+0x39c
#21  0x7f7eb4bf943c in App::Document::recomputeFeature(App::DocumentObject*, bool) from /opt/FreeCAD/lib/libFreeCADApp.so+0x12c
#22  0x7f7eb4c60d77 in App::DocumentObjectPy::recompute(_object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x57
#23  0x7f7eb4c60e83 in App::DocumentObjectPy::staticCallback_recompute(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x23
#24  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541) [0x7f7eb4141091]
#25  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#26  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e76d) [0x7f7eb41fc76d]
#27  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#28  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x36f7) [0x7f7eb413c247]
#29  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#30  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e76d) [0x7f7eb41fc76d]
#31  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#32  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x36f7) [0x7f7eb413c247]
#33  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#34  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#35  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#36  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#37  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#38  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e670) [0x7f7eb41fc670]
#39  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#40  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193dfc) [0x7f7eb4251dfc]
#41  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#42  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7eb42a76c7]
#43  0x7f7eac7d0309 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /usr/lib/x86_64-linux-gnu/libpyside-python2.7.so.1.2+0x79
#44  /usr/lib/x86_64-linux-gnu/libpyside-python2.7.so.1.2(+0x169c9) [0x7f7eac7d79c9]
#45  0x7f7ead2ff78e in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x5fe
#46  /usr/lib/python2.7/dist-packages/PySide/QtCore.so(+0x1a17a3) [0x7f7e8d27b7a3]
#47  0x7f7ead305253 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0xa3
#48  /usr/lib/python2.7/dist-packages/PySide/QtCore.so(+0x1a1323) [0x7f7e8d27b323]
#49  0x7f7eaddb154c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#50  0x7f7eaddb852c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#51  0x7f7eb5574441 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
[... cut ...]
#65  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f7eabba52e1]
#66  freecad-git(_start+0x2a) [0x55a02f7e2a0a]
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16513 (Git)
Build type: Release
Branch: LinkStage3
Hash: 10f21ea81873b36a491b46cd7d20f23647f83c58
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachment 

I can privately send if needed.

ceremcem, 2019-04-13:
So far so good. This is a very useful feature. Many thanks.

closed 2019-04-13 by ceremcem