realthunder/FreeCAD_assembly3 #207: "Discard" button is dangerous

ceremcem, 2019-03-27:
Discarding an expression in sketcher constraint is dangerous because it completely removes the expression without a way to undo or cancel: 

[image]

closed 2019-03-29 by realthunder

ceremcem, 2019-03-29:
Nice! Thanks.

ceremcem, 2019-08-06:
This issue repeats in current version: 

[image]

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18681 (Git)
Build type: Release
Branch: LinkStage3
Hash: 6a6302d761ee5db77d5f1042311a5663a42d7798
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-08-07:
It is not that dangerous, because you can undo it. I have just added a commit to auto undo it after canceling. I have added a few patches to improve the undo/redo system. Let know if you find any operation that cannot be undone.

ceremcem, 2019-08-07:
That's true, I didn't try to undo after observing unexpected behavior of "Cancel" button. Anyway, current behavior is very user friendly. Thanks.