realthunder/FreeCAD_assembly3 #208: Definition of "action" in undo/redo

ceremcem, 2019-03-27:
Undo/redo does not behave as I expected, so I generally try to avoid using it. 

### Expected 

Undo/redo should behave as if: 
* I save all documents (current document and its dependencies). 
* Zip it with current timestamp (`myproject.t0.zip`)
* Do a single change
* Save everything and zip with current timestamp (`myproject.t1.zip`)
* Close all documents 
* Unzip `myproject.t0.zip` for undo
* Unzip `myproject.t1.zip` for redo

### Observed

Undo/redo does not cover property changes. For example:

* Edit a sketcher constraint
* Undo
* You won't be able to restore the previous constraint contents

[image]

realthunder, 2019-03-28:
The steps you expected are for making backups, not undo/redo, because of the prohibiting performance penalty. Saving/Restoring documents in FC is expensive. The actual implementation is kind of complicated in FC. I have added a few patches to help improve transaction in FC, but ultimately it still relies on each individual workbench to setup transaction at proper moment. And the sketcher is a bad example of doing this. I don't think there is any general solution to this problem.

ceremcem, 2019-03-29:
> I don't think there is any general solution to this problem.

I'm not sure if there is a general solution or not. My current approach in my app is really taking full backups of whole project in every change. This obviously introduces huge undo/redo history db size on disk and costs some CPU time, but this is the most robust way to go. In the future, I'll implement a diff algorithm to save last N actions inside the project file without impacting file size. 

Besides, undo/redo is very problematic also in Solidworks, according to my knowledge. 

What I understand is that every workbench is responsible for what to include into the undo history. If that's the case, it's a source of anxiety and the correct way to go is really having full backups periodically.

ceremcem, 2019-04-02:
Following actions are **excluded** from the undo/redo history: 

* Sketcher constraints changes 
* Spreadsheet "Paste" actions
(to be continued)

closed 2019-04-02 by ceremcem