realthunder/FreeCAD_assembly3 #212: Add timeout for each expression

ceremcem, 2019-04-02:
We may accidentally write an infinite loop in an expression (like I did a few minutes ago). FreeCAD freezes in such a case, which can be expected. However, we are forced to kill the app in those cases and this leads loss of unsaved work. 

If we could assign a specific timeout to a global `TIMEOUT` variable for each expression, we can both handle such programmer errors and fine tune the value for heavy computation required operations. A default value of `10 seconds` is more than enough for `TIMEOUT` variable.

realthunder, 2019-04-17:
You can now press 'ESC' to abort spreadsheet recomputation.

closed 2019-04-17 by realthunder

ceremcem, 2019-04-17:
Nice. Seems more handy way than a TIMEOUT approach.