realthunder/FreeCAD_assembly3 #214: Incorrect error in expression engine

ceremcem, 2019-04-02:
In the [M3x16-bolts.fcstd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3036164/M3x16-bolts.fcstd.zip)
 document, the following error is shown in Report View when the document is first opened: 

```
<Spreadsheet> Sheet.cpp(1387): Failed to restore M3x16_bolts#Spreadsheet: One or more cells failed contains errors.
```

However, there is no error in the document. When I manually click the "Recompute" button, it successfully computes the values. 

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16495 (Git)
Build type: Release
Branch: LinkStage3
Hash: 53da7243ec1157c02911f57e3d59706b08a3c1ed
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2019-04-03:
[image]

First spreadsheet depends on second and third spreadsheets. Is it possible that the dependencies may be unavailable at the first load because of the spreadsheet orders?

realthunder, 2019-04-04:
Object reference inside function has no dependency tracking, and this is by design. I just added a patch to warn user when such function is run. In almost all cases, it is better to pass the object reference as argument to the function, something like,

```python
def Lookup(key, obj):
    #@pybegin
    row_num = 1
    while True :
        try:
            _key = obj.get('A' + str(row_num))
            if str(_key) == str(key):
                value = obj.get('B' + str(row_num))
                break
        except:
            raise "No value can be found for key " + str(key)

        row_num += 1
```

And you call it by `.Lookup(.B2, <<drill_table>>._self)`

ceremcem, 2019-04-04:
Okay, now it's nicely working. Here is the latest bolt model: 
[M3x16-bolts.fcstd.remove-me.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3045337/M3x16-bolts.fcstd.remove-me.zip)

Now, if you need a `M4x20mm` model, just duplicate this model with the name "`M4x20-bolts.fcstd`" and open. IMHO, this is a killer feature of Asm3.

closed 2019-04-04 by ceremcem