realthunder/FreeCAD_assembly3 #215: Unexpected identifier error

ceremcem, 2019-04-02:
The following expression throws the following error: 

[image]

Where I wouldn't expect such an error.

realthunder, 2019-04-02:
You are using the wrong quote. Should be `<<A2>>`.

ceremcem, 2019-04-02:
It's confusing. How would I possibly concatenate row number with the column letter if I were to use `<<...>>`? Following will not work: 

```
=._self.get(<<A>> + str(2))
```

realthunder, 2019-04-02:
Yeah, I know. I hate that quoting style, too. Unfortunately, that's how upstream expression works. They use `'` for unit foot, or is it inch?

Your error above is because `str` is not an expression-built-in function. It is a Python built-in, which can be access by `._py` pseudo property. So `<<A>> + _py.str(2)` will work.

realthunder, 2019-04-02:
Here is an idea. I will add an property to `Spreadsheet` to turn on python mode by default.

ceremcem, 2019-04-03:
Now I got the root of my confusion. Is there 2 distinct modes in expression engine now; "Default FreeCAD Expressions" and "Python Expressions"; or is there only an "Extended Expression Engine", which *tries* to be compatible with "Default FreeCAD Expressions"? What I understand is there is only the extended version now. 

> Here is an idea. I will add an property to Spreadsheet to turn on python mode by default.

That would be nice for most of my use cases. However, how can someone else (another developer) switch to normal mode once it'd been set to "Python Mode" beforehand?

My proposal would be adding a Shebang-like starting flag after `=` sign. For example, following code might be interpreted as Python code: 

```
=#py str(3)
```

We have to type this additional `#py` in every cell, but it's easy to type anyway.

realthunder, 2019-04-03:
There is only the `Extended Expression Engine`. By `Extended`, I mean it supports an extended syntax of the original. The original syntax has several conflicts with the new Python-like syntax. In order to maintain backward compatibility, so I introduce some pseudo statements for switching the mode, which is documented [here](../wiki/Expression-and-Spreadsheet#python-syntax-mode).

And yes, you can use the `Shebang-like` flag as shown below. It allows you to use the Python quote directly, but you still cannot directly reference the Python built-ins, which involves name lookup, which is only possible within some function scope. I may lift that restriction in the future.

```
=#@pybegin; 'a' + ._py.str(3)
```

> how can someone else (another developer) switch to normal mode once it'd been set to "Python Mode" beforehand?

I intended it to be a simple boolean property attached to the spreadsheet. So each spreadsheet can have different default mode. And of course, the mode can be overridden using `#@pyend` as normal.

ceremcem, 2019-04-03:
> I intended it to be a simple boolean property attached to the spreadsheet.

My scenario was something like that:
1. I (dev_a) prefer using the default Python mode. 
2. I created a model and created a new spreadsheet, labeled as `foo`. 
3. I set the "Use Python mode by default" flag to true for spreadsheet `foo`. 
4. I've finished working on my model. 
5. Another developer (dev_b) wanted to modify my model and start editing spreadsheet `foo`. 
6. dev_b wants to use the upstream-like syntax. 

If dev_b wants to switch the "Use Python mode by default" flag off for `foo`, dev_a's expressions will be broken. So you say the dev_b may escape his upstream-like syntax with `=#@pyend; ...`? That would be okay too. 

> And yes, you can use the Shebang-like flag as shown below

Nice, that's something like a hidden treasure :) I think I'll stick to that when I need to use Python mode. It's verbose, but declarative. 

>  so I introduce some pseudo statements for switching the mode

Honestly, I didn't understand how to use them. In function definition there was a `#@pystart` line in the examples, so I was just replicating the usage without understanding what it's there for. This section is not clear, but naturally I can't tell what's unclear before understanding a little bit about it.

realthunder, 2019-04-03:
The feature is implemented, also the name lookup outside function scope. When `Spreadsheet.PythonMode = True`, you can now do

```
'a' + str(1)
```

ceremcem, 2019-04-03:
Is it correct to demonstrate the inclusion diagram as follows: 

[image]

Here, 
* `#1`: [Python-syntax-mode](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#python-syntax-mode)
* `#2`: The default mode while entering an expression, unless
    * `Python Mode` is set to `True` for a spreadsheet
    * `#@pybegin;` isn't prepended to an expression 
* `#3`: None?
* `#4`: [not-implemented](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#not-implemented), [security-related](https://github.com/realthunder/FreeCAD_assembly3/wiki/Expression-and-Spreadsheet#security-concern)

realthunder, 2019-04-04:
Nice! That's about it. You can submit a PR to modify that wiki if you want.

ceremcem, 2019-04-04:
Thanks.

> You can submit a PR

Here is the applied branch: https://github.com/ceremcem/freecad-asm3-wiki/tree/add-overall-map You need to manually pull from my branch, AFAIK. BTW, If you could maintain the wiki as if it was a regular repository, it would be far easier to submit a PR. 

(Note that I'm not quite sure about the appearance, please feel free to criticise about the fine tuning)

closed 2019-04-04 by ceremcem