realthunder/FreeCAD_assembly3 #216: Broken model

ceremcem, 2019-04-03:
Following model is broken and might be related with #214: 

[M3x16-bolts.fcstd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3038197/M3x16-bolts.fcstd.zip)

I'm no longer able to see the properties of any object in the property window: 

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16495 (Git)
Build type: Release
Branch: LinkStage3
Hash: 53da7243ec1157c02911f57e3d59706b08a3c1ed
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-04-03:
Please sync the branch and try again. There is a few expression error in the spreadsheet, but nothing seems broken when open in my build.

ceremcem, 2019-04-03:
I synced the branch and nothing went wrong on my side either. Properties are shown as expected.

closed 2019-04-03 by ceremcem