realthunder/FreeCAD_assembly3 #221: Object property expressions are not updated upon spreadsheet update

ceremcem, 2019-04-05:
# Reproduction 

* Download this model: [M3x16-bolts.fcstd.remove-me.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3045337/M3x16-bolts.fcstd.remove-me.zip)
* Open the model and see the `<<dimensions>>.B9` cell is set to `16mm`, which is also aliased to `length`. 
* See the `allen.Parts.hole-cut002.Height` property is set to `<<dimensions>>.length`, which is `16mm` at the moment and is correct. 
* Duplicate this model and rename the copy as `M4x40-bolts.fcstd`, and open it. 
* See the `<<dimensions>>.B9` cell is calculated as `40mm`. 
* See the `allen.Parts.hole-cut002.Height` is **still** set to `16mm`
* Click this property, press `=` to enter the expression mode
* See the expression is correctly displayed as `<<dimensions>>.length` and *is correctly calculated* as `40mm`
* Press enter
* See the `allen.Parts.hole-cut002.Height` is now set to `40mm`. 

# Screencast 

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16498 (Git)
Build type: Release
Branch: LinkStage3
Hash: cc4acda7877b84b640dc87c4d8a260aaf8d2ca59
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-04-08:
This is not really a bug. The spreadsheet only stores the expression, not the evaluated results, which can be arbitrary Python object that may not be persistent. Therefore, it cannot detect result change triggered by anything outside the document, and right now, the document label is consider as such, because it is derived from the document filename which is not stored as part of the document data. 

The property expression binding, on the contrary, does store the evaluation result as if it was entered manually. However, there is no expression evaluation and result comparison at document loading time, as this may be potentially expensive operation. Besides, there may not by any practical comparison operation available on certain type of result, such as a geometrical shape. BTW, the spreadsheet does evaluate the expression on document open (since it does store the result), so you should avoid writing expensive expression inside it. A function definition expression is (usually) not considered as expensive, regardless of what complex logic the function perform, since the function is not executed during expression evaluation time.

I think there is a larger problem of your intended use case. If you are using spreadsheet for coding purpose, then code duplication should not be encouraged. Imagine you find some problem with the code, and want to fix it. For your use case, I would define function inside spreadsheet to directly generate the shape. And bind that shape to a 'Part::Feature'. 

I actually reworked expression engine with this type of use case in mind, specifically, use `cadquery` to build a chip component library. I have used some libraries like [this](https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator) before I worked on FC. One disadvantage is that there is no easy way to change parameters inside those cadquery scripts. You'll have to modify the source in case your want customization. Now you can define shape generating functions inside some spreadsheet and consider it as library. To use the library, create another spreadsheet in your assembly file, and call the function in the library with desired parameters.

PS. it seems that cadquery is moving away from FC. That's unfortunate. I think we can directly bundle a copy of cadquery module inside FC while it still works.

ceremcem, 2019-04-08:
> Therefore, it cannot detect result change triggered by anything outside the document

If that's true, spreadsheet wouldn't update `<<dimensions>>.B9` cell from `16mm` to `40mm` on document load. If spreadsheet does know it needs to update a cell, it should trigger the dependants accordingly. 

> BTW, the spreadsheet does evaluate the expression on document open (since it does store the result), 

You mean "since it doesn't store the result", right?

> However, there is no expression evaluation and result comparison at document loading time, as this may be potentially expensive operation.

Is the following logic still expensive: 

1. Export a spreadsheet's current state (the evaluated values) (for example, in CSV format).
2. Calculate the MD5 sum on document close.
3. Store this checksum as a part of the spreadsheet. 
4. Compare the checksums to see if there is a change with the evaluated values on document open. 
5. Trigger a change if checksums differ. 

Anyway, I think there should be a way to re-evaluate the property expression bindings.

realthunder, 2019-04-08:
> If that's true, spreadsheet wouldn't update `<<dimensions>>.B9` cell from `16mm` to `40mm` on document load. If spreadsheet does know it needs to update a cell, it should trigger the dependants accordingly.

Spreadsheet evaluates the expression on opening, but since it does not store the result, it has nothing to compare to.

> You mean "since it doesn't store the result", right?

Right, my bad.

> Is the following logic still expensive:

Think about a function object, or a geometry shape. Sometimes, it does not make sense or practical to compare the result. We can only compare the expression that generates the result, plus any input parameters. So if the input is part of the document data, then we can compare and detect change. But in your case, the document label is not considered as part of the document data. Dependency tracking only works on object property, not document property.

> Anyway, I think there should be a way to re-evaluate the property expression bindings.

You can always manually touch the property (right click the property, select 'Show All', and right click again, select 'Touch'). But of course, we are talking about automatic change detection here, which relies on expression dependency tracking. Right now, that mechanism does not work on document property. It is not impossible to add that functionality, but it is not trivial either. Besides, I do have doubts about your current approach. What do you think about my shape generation idea. You don't have to use cadquery, the Part module is already exposed by `._part`

ceremcem, 2019-04-09:
> Besides, I do have doubts about your current approach. 

First of all, let me clarify the confusion here: 

* When document loads, spreadsheet cells are correctly updated. 
* When spreadsheet updates, object property expressions are correctly updated (while document is opened)

So *if spreadsheet cells are updated on document load, object property expressions are expected to be updated as well*. Anything other than this is confusing.

#### Approach explained

Think that document itself is the `class`, copies are the `instance`s and their initialization parameters are passed by document names. This is an absolutely charming feature because it both eases maintainability and speeds up variant generation. Besides, this approach perfectly fits diverging a model in an agile development pace because every instance is a standalone file so if a model is diverged (this happens suddenly), just move it to another folder and keep using it.

Let's drop the full automatic approach, which is something I was already using via [a macro](https://github.com/ceremcem/assembly-dimensions, see `#! document-name` section.). Very same approach can also be achieved by manually entering the document name into a cell. If I can achieve what I want by manually inserting the document name into a cell (or by a macro) and I am currently (with extended expression engine) able to calculate the document name within a cell, why shouldn't I expect this very last step (object property expression update part) performed automatically? 

If this support is omitted, it will be both unfortunate and confusing. It's up to you to add this support. 

> One disadvantage is that there is no easy way to change parameters inside those cadquery scripts. You'll have to modify the source in case your want customization.

Same applies to FreeCAD documents. Only the "Parameters passed by document name" approach overcomes this issue. 

> Now you can define shape generating functions inside some spreadsheet and consider it as library.
> You don't have to use cadquery, the Part module is already exposed by ._part

It would be far easier to understand if there were an example for this use case. Would you prepare an example that dynamically generates a cylinder by getting the height parameter from a spreadsheet cell?

realthunder, 2019-04-11:
> Let's drop the full automatic approach, which is something I was already using via [a macro](https://github.com/ceremcem/assembly-dimensions)

I have just added a manual approach. You need to know beforehand which property expression binding depends on untracked change source, such as the document label. In you attached file, I think all expression bindings depend on it? You need to manually turn on the `EvalOnRestore` option on that property. 

In the property editor, right click anywhere, and select `Show all`. Then for each property you want to evaluate the expression on restore, right click the property and select `EvalOnRestore`. Save the document, rename it and reopen it. Any affected objects will be touched. You need to manually recompute the document to get a complete update. Be noted that not all property types support meaningful comparison operation.

> It would be far easier to understand if there were an example for this use case.

I'll post some example later.

realthunder, 2019-04-22:
Here is a demo of expression generationed parametric shapes. Create a new
document (should have default name `Unnamed`), and then create an empty
spreadsheet. Copy the following scripts into clipboard, and right click the
spreadsheet item in the tree view, select `Expression actions -> Paste`. Then
recompute the document.

You will find three buttons in the spreadsheet. The first one creates a simple
dumb shape generated from calling the `makeCylinder()` function in cell `A2`.

The second button creates a `Part::FeaturePython` with two properties `Height`
and `Radius`. The `Shape` is generated using expression binding to
`Spreadsheet.makeCylinder()`. When you change either `Height` or `Radius`, the
shape will be regenerated as expected. This object depends on the spreadsheet,
hence the name `attached`.

The third button does mostly the same as previous, except that it copies the
expression from spreadhsheet cell `A2` (i.e. the code that generates the shape)
into its shape expression binding, and thus making this object self contained,
hence the name `detached`.

```python
##@@ A1 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="MakeSimpleObject" editMode="1" />
def MakeSimpleObject():
    #@pybegin
    pragma('nowarn', 0) #disable warning
    ._self.Document.addObject('Part::Feature', 'simple').Shape = .makeCylinder()

##@@ A2 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="makeCylinder" />
def makeCylinder(radius=.Radius, height=.Height, pla=None):
    #@pybegin
    shape = ._part.makeCylinder(radius, height)
    if pla:
        shape.Placement = pla

    return shape

##@@ A3 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@
<<Height>>

##@@ A4 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@
<<Radius>>

##@@ B1 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="MakeAttachedObject" editMode="1" />
def MakeAttachedObject():
    #@pybegin
    pragma('nowarn', 0)
    obj = ._self.Document.addObject('Part::FeaturePython', 'attached')
    obj.addProperty('App::PropertyLength', 'Radius')
    obj.addProperty('App::PropertyLength', 'Height')
    obj.Radius = .Radius
    obj.Height = .Height
    obj.ViewObject.Proxy = True
    obj.setExpression('Shape', ._self.Name + '.makeCylinder(height=.Height,radius=.Radius,pla=.Placement)')

##@@ B3 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="Height" />
(10mm)

##@@ B4 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="Radius" />
(5mm)

##@@ C1 Unnamed#Spreadsheet.cells (Spreadsheet)
##@@<Cell alias="MakeDetachedObject" editMode="1" />
def MakeDetachedObject():
    #@pybegin
    pragma('nowarn', 0)
    obj = ._self.Document.addObject('Part::FeaturePython', 'detached')
    obj.addProperty('App::PropertyLength', 'Height')
    obj.addProperty('App::PropertyLength', 'Radius')
    obj.Height = .Height
    obj.Radius = .Radius
    obj.ViewObject.Proxy = True

    # Unlink above, we are coping the expression code (using getContents())
    # instead of calling it.  The code defines the function named
    # `makeCyliner()`, we then immediate calls this function inside the same
    # expression binding.

    obj.setExpression('Shape', ._self.getContents('makeCylinder')[1:] + '\nmakeCylinder(pla=.Placement)')
```