realthunder/FreeCAD_assembly3 #223: Exporting mesh from assembly produces empty mesh

bgamari, 2019-04-09:
Exporting a mesh (e.g. an STL via `File -> Export`) from an Assembly or Parts group produces an empty mesh file.

# Reproduction

1. Create a new Part, add some geometry to it.
2. Create a new Assembly
3. Drag the Part into the assembly
4. Select the Assembly and export it to a mesh file (e.g. STL) via `File -> Export`.
5. Open the STL file and note that it is empty.

realthunder, 2019-04-09:
I'll add support for that soon. Meanwhile, you can work around this by setting the `BuildShape` property of the assembly to `Compound`, and then export will work.

bgamari, 2019-04-09:
Ahh, great. Thanks @realthunder!

luzpaz, 2021-02-21:
@realthunder was this implemented?

realthunder, 2021-02-22:
Yes, it is implemented.

closed 2021-02-22 by realthunder