realthunder/FreeCAD_assembly3 #224: Grouping makes FreeCAD crash

ceremcem, 2019-04-10:
# Screencast 

[image]

# Backtrace 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f7eabbb8060]
#1  0x7f7eb4c4c445 in App::ExtensionContainer::hasExtension(Base::Type, bool) const from /opt/FreeCAD/lib/libFreeCADApp.so+0x15
#2  0x7f7eb4d2e836 in App::LinkBaseExtension::flattenSubname(char const*) const from /opt/FreeCAD/lib/libFreeCADApp.so+0xa6
#3  0x7f7eb4d427b1 in App::LinkBaseExtensionPy::flattenSubname(_object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x41
#4  0x7f7eb4d42ad3 in App::LinkBaseExtensionPy::staticCallback_flattenSubname(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x23
#5  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541) [0x7f7eb4141091]
#6  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f7eb413f390]
#7  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#9  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#10  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e670) [0x7f7eb41fc670]
#11  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#12  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193dfc) [0x7f7eb4251dfc]
#13  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#14  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7eb42a76c7]
#15  0x7f7eb4cf58cb in App::FeaturePythonImp::execute() from /opt/FreeCAD/lib/libFreeCADApp.so+0x18b
#16  0x7f7e8b6dd085 in App::FeaturePythonT<Part::Feature>::execute() from /opt/FreeCAD/lib/Part.so+0x15
#17  0x7f7eb4c41e19 in App::DocumentObject::recompute() from /opt/FreeCAD/lib/libFreeCADApp.so+0x79
#18  0x7f7e8b6cf0db in Part::Feature::recompute() from /opt/FreeCAD/lib/Part.so+0xb
#19  0x7f7eb4be055d in App::Document::_recomputeFeature(App::DocumentObject*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x2ed
#20  0x7f7eb4bf8bac in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /opt/FreeCAD/lib/libFreeCADApp.so+0x39c
#21  0x7f7eb4bf943c in App::Document::recomputeFeature(App::DocumentObject*, bool) from /opt/FreeCAD/lib/libFreeCADApp.so+0x12c
#22  0x7f7eb4c60d77 in App::DocumentObjectPy::recompute(_object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x57
#23  0x7f7eb4c60e83 in App::DocumentObjectPy::staticCallback_recompute(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x23
#24  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541) [0x7f7eb4141091]
#25  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#26  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e76d) [0x7f7eb41fc76d]
#27  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#28  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x36f7) [0x7f7eb413c247]
#29  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#30  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e76d) [0x7f7eb41fc76d]
#31  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#32  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x36f7) [0x7f7eb413c247]
#33  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#34  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#35  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#36  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x670d) [0x7f7eb413f25d]
#37  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f7eb42a829c]
#38  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x13e670) [0x7f7eb41fc670]
#39  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#40  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x193dfc) [0x7f7eb4251dfc]
#41  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x43) [0x7f7eb41945c3]
#42  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7f7eb42a76c7]
#43  0x7f7eac7d0309 in PySide::SignalManager::callPythonMetaMethod(QMetaMethod const&, void**, _object*, bool) from /usr/lib/x86_64-linux-gnu/libpyside-python2.7.so.1.2+0x79
#44  /usr/lib/x86_64-linux-gnu/libpyside-python2.7.so.1.2(+0x169c9) [0x7f7eac7d79c9]
#45  0x7f7ead2ff78e in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x5fe
#46  /usr/lib/python2.7/dist-packages/PySide/QtCore.so(+0x1a17a3) [0x7f7e8d27b7a3]
#47  0x7f7ead305253 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0xa3
#48  /usr/lib/python2.7/dist-packages/PySide/QtCore.so(+0x1a1323) [0x7f7e8d27b323]
#49  0x7f7eaddb154c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#50  0x7f7eaddb852c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#51  0x7f7eb5574441 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#52  0x7f7ead2eaf1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#53  /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1bdce0) [0x7f7ead31dce0]
#54  /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1baee1) [0x7f7ead31aee1]
#55  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f7ea48807f7]
#56  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f7ea4880a60]
#57  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f7ea4880b0c]
#58  0x7f7ead31b854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#59  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f7eade5b5d6]
#60  0x7f7ead2e97ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#61  0x7f7ead2e9b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#62  0x7f7ead2efbd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#63  0x7f7eb54ee49a in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x18da
#64  freecad-git(main+0x862) [0x55a02f7e17b2]
#65  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f7eabba52e1]
#66  freecad-git(_start+0x2a) [0x55a02f7e2a0a]
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16513 (Git)
Build type: Release
Branch: LinkStage3
Hash: 10f21ea81873b36a491b46cd7d20f23647f83c58
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachment 

I can privately send if needed.

realthunder, 2019-04-10:
Should be fixed now, Please sync and check.

closed 2019-04-10 by realthunder

ceremcem, 2019-04-10:
I synced, FreeCAD does not crash now but there is some other error occurs. 
[image]

Those errors instantly occurred when I clicked the "Group" button. 

I'm handing over the model via FreeCAD forum PM. My expectation is like so: 

- [ ] When a component (object) (actual one or a link) is clicked, the group should cover the component. 
- [ ] Another component should be dragged into the group
- [ ] Any object (whether under `Elements` or `Parts` section in an `Assembly`) should be able to drag outside the group (currently not possible)
- [ ] Feature: It would be convenient to take an element outside of current group (possibly by "right click/move upper level")
- [ ] Feature: Ungroup objects: A group container should be able to be deleted without deleting the objects inside it (just like deleting a Fusion where the operand objects left intact)