realthunder/FreeCAD_assembly3 #229: PointInPlane constraint works wrong

ceremcem, 2019-04-12:
When a line is used as a Point, the mid point is used as the Point. This means, when a line and a plane (where the plane normal is parallel to the line) is used in PointInPlane constraint, the plane should be placed in the middle of the line. However, currently one end of the line is considered as the Point. That behavior is correct in `commit: 39c32719884d1a56c569cd38da56eee96b8ca1b8`. (sorry, I've purged all snapshots some time ago, only this very old snapshot is left)

# Version 
``` 
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16534 (Git)
Build type: Release
Branch: LinkStage3
Hash: b02b8c3d03e208c0dfdf84155f728b89713e5a61
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Screencast 

[image]

# Attachment 

[point-in-plane-wrong.fcstd.remove-me.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3073178/point-in-plane-wrong.fcstd.remove-me.zip)

realthunder, 2019-04-12:
I was puzzled by the inconsistent constraint error too, but haven't got time to look deep into it. Good finding! Saved me lot of time!

closed 2019-04-12 by realthunder

ceremcem, 2019-04-12:
The speed of fix for the issue might be the description of the speed of light. Can't believe this.

commit cad2e8f8 (2019-04-12): utils: fix getElementPos() one line edge

Fixes #229
Related #225