realthunder/FreeCAD_assembly3 #230: Failing to create element from linked document

ceremcem, 2019-04-12:
# Priority 

Blocker 

# Description 

I can't create element from a geometry of a linked document. 

# Screencast 

[image]

# Version

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16536 (Git)
Build type: Release
Branch: LinkStage3
Hash: e9c60d4d8d72608fe897000c9658950419774b63
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments

[test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3075213/test.zip)

realthunder, 2019-04-13:
Oops! Another careless mistake on recent commits. Shame on me!

realthunder, 2019-04-13:
You can fix the broken link by manually drag and drop the head_start element on to it.

closed 2019-04-13 by realthunder

ceremcem, 2019-04-13:
Thanks! (We really need that test suite)

realthunder <notifications@github.com>, 13 Nis 2019 Cmt, 03:15 tarihinde
şunu yazdı:

commit 49812ba3 (2019-04-13): assembly: fix AsmElement.make()

Fixes #230