realthunder/FreeCAD_assembly3 #234: Relation can not be found if object is a remote link and inside an Asm Group

ceremcem, 2019-04-14:
# Description 

The "Select object in relation group" button throws the following exception if the object is a link from a remote file and placed under a group:

```
34168.8 Selection.cpp(1560): Sub-object Unnamed#Assembly.Parts.3.Relation001. not found
```

# Screencast 

[image]

# Attachments 

[group-breaks-linked-relation.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3077081/group-breaks-linked-relation.zip)

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16536 (Git)
Build type: Release
Branch: LinkStage3
Hash: e9c60d4d8d72608fe897000c9658950419774b63
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Workaround 

Temporarily move the link outside of the group.

realthunder, 2019-04-28:
Please sync my branch and try again. I cannot reproduce this problem now.

ceremcem, 2019-04-28:
Here is the current exception on the Report View:

[image]

```
Running the Python command 'asm3CmdGotoRelation' failed:
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 682, in Activated
    AsmRelationGroup.gotoRelation(AsmCmdMove._moveInfo)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2638, in gotoRelation
    objName(moveInfo.SelObj),moveInfo.SelSubname))

Invalid selection button_driver#Assembly.Parts.Link.Parts.Group004.Link010.Running the Python command 'asm3CmdGotoRelation' failed:
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 682, in Activated
    AsmRelationGroup.gotoRelation(AsmCmdMove._moveInfo)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2638, in gotoRelation
    objName(moveInfo.SelObj),moveInfo.SelSubname))

Invalid selection button_driver#Assembly.Parts.Link.Parts.Group004.Link010.
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16597 (Git)
Build type: Release
Branch: LinkStage3
Hash: 6d40419fac41708768bb57062d74ab08a2d138d0
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

closed 2019-04-30 by realthunder

realthunder, 2019-04-30:
Should be working now. I have also changed GoToRelation behavior. Previously, if the selection resides in a multi-hierarchy assembly, it always go to the relation of the top most assembly. Now, if the selection is a constraint or element, it goes to the relation inside the bottom assembly.

ceremcem, 2019-04-30:
Unfortunately no, it's still throwing exception: 

```
Running the Python command 'asm3CmdGotoRelation' failed:
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/gui.py", line 682, in Activated
    AsmRelationGroup.gotoRelation(AsmCmdMove._moveInfo)
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 2686, in gotoRelation
    subs[-3] = '3'

list assignment index out of range
```

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16600 (Git)
Build type: Release
Branch: LinkStage3
Hash: ed49103b4671a7621e94455a15b78949a8dee20d
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

commit ddd3250f (2019-04-30): assembly: fix gotToRelation on grouped parts

Fixes #234