realthunder/FreeCAD_assembly3 #236: Can we depend on elements to track the ID of a face

ceremcem, 2019-04-16:
Fully deterministic ID generation for `Face`s, `Edge`s and `Vertex`es of an object is impossible and the enumeration starts from beginning after any geometry change. 

However, I noticed that the `Element`s does track its `Linked Object` *as much as possible*, which exactly fits my needs. **Question 1/2:** Can we depend on this behavior? 

# Intention and problem

I'm trying to provide a solution to https://github.com/shaise/FreeCAD_SheetMetal/issues/40, basically a script that will re-unfold a processed object after every geometry change by using a manually provided face ID, such as `Gui.Selection.addSelection('Unnamed','Cut','Face11')`. 

[image]

Even though we don't touch the geometry of the selected face intact with a careful development, ID's may change: 

[image]

# Possible approach

1. Add the object into an Assembly Container
2. Create an `Element` for the face
3. Use the `LinkedObject` property to determine the ID of the previous face: 

  ```
  >>> App.ActiveDocument.getObjectsByLabel('unfold_face')[0].LinkedObject
  (<Part::PartFeature>, ';#11f;:T559:4:F.Face12')
  ```

**Question 2/2:** If it's a valid usage, how can I properly convert `';#11f;:T559:4:F.Face12'` to `'Assembly','Parts.Cut.Face12'`?

ceremcem, 2020-06-15:
@realthunder Could you please take a look at this? 

Briefly, I'm asking how to add the face that an `Element` points, to the selection via `Gui.Selection.addSelection(...)`.

realthunder, 2020-06-15:
It is a bit complicated. `Element.LinkedObject` stores the linked object and subname. But an `Element` can point to other `Element` down the hierarchy. You can use `Element.Proxy.getElementSubName(True)` to recursively resolve the subname to the actual geometry. The subname is relative to the `PartGroup` of the same assembly that contains this `Element`

So assuming the `Element` you want is named simply as `Element`, then

```python
element = App.ActiveDocument.Element.Proxy
partGroup = element.getAssembly().getPartGroup()
subname = element.getElementSubname(True)
Gui.Selection.addSelection(partGroup, subname)
```
There are two complications. 1) The returned subname is flattened, meaning that any plain group you created will not be in it. So tree view sync may not work.  2) The element you want may not in the current document. It may be external linked.

Alternatively, you may want to try using two asm3 commands. You need to first select the element you want in the tree view, then run
```python
Gui.runCommand('Std_LinkSelectLinked',0)
```
This is equivalent to click the `Select linked object` toolbar button. To go to the deepest link
```python
Gui.runCommand('Std_LinkSelectLinked',1)
```

ceremcem, 2020-06-30:
The problem with `Gui.runCommand('Std_LinkSelectLinked',0)` or `..., 1)` is that this command selects the whole object, not the face. 

The example for the element `Element` works nicely on Python console. However, it wasn't working in Macro, for multiple documents. Turns out it works only for the active document. Here is the working version for anyone needs it: 

```py
doc_name="driver_base"
elem_label="wall_unfold_face"
# ------------------------------------------------------------
Gui.ActiveDocument=Gui.getDocument(doc_name)
App.ActiveDocument=App.getDocument(doc_name)
App.setActiveDocument(App.ActiveDocument.Name) # Interesting. Why do we need to assign this manually?
doc=App.ActiveDocument
element = doc.getObjectsByLabel(elem_label)[0].Proxy
partGroup = element.getAssembly().getPartGroup()
subname = element.getElementSubname(True)
Gui.Selection.addSelection(partGroup, subname)
```

closed 2020-06-30 by ceremcem