realthunder/FreeCAD_assembly3 #238: Path WB doesn't work

ceremcem, 2019-04-20:
When I switch to Path WB, the following error message is displayed: 

```
/opt/FreeCAD/lib/libocc.so.0: undefined symbol: _ZNK12MMgt_TShared6DeleteEv
```

#  Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16548 (Git)
Build type: Release
Branch: LinkStage3
Hash: 3559163f7db84b9a2958fd58e76c3481ffceab16
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-04-20:
libocc.so is used by FEM Netgen. I didn't enable this module on my system. Maybe you can turn that off if you don't need it. Find the CMakeCache.text in your build directory. Find `BUILD_FEM_NETGEN` inside, and set it to OFF, and compile. Not sure why suddenly you have this problem.

ceremcem, 2019-04-20:
> Maybe you can turn that off if you don't need it.

Yeah, I think I don't need it at all. I even don't know why it is there for. 

> Not sure why suddenly you have this problem.

That's interesting, yes. AFAIR, the last time I tried Path WB at the time of https://github.com/realthunder/FreeCAD/issues/7 submitted and I was using it without such an exception.  Anyway, disabling the FEM Netgen won't effect my workflow at all.

ceremcem, 2019-04-24:
The above exception is gone after compiling FreeCAD without Netgen support.

closed 2019-04-24 by ceremcem