realthunder/FreeCAD_assembly3 #240: FreeCAD crash

ceremcem, 2019-04-27:
# Last actions 

I did nothing but a "Recompute" after opening my project. First an error is thrown while complaining about a missing mapped element: 

```
<PropertyLinks> PropertyLinks.cpp(367): button_driver#Element.LinkedObject missing element reference button_driver#Cut ;#24c1;:M;CUT;:T19:5:F.Face21
<Exception> TopoShapeEx.cpp(518): Mapped element not found: ;#24c1;:M;CUT;:T19:5:F.Face21
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 515, in execute
    self.getElementSubname())
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1153, in getElementInfo
    shape = utils.getElementShape((part,subname))
  File "/opt/FreeCAD/Ext/freecad/asm3/utils.py", line 89, in getElementShape
    transform=transform,noElementMap=noElementMap)
<type 'exceptions.RuntimeError'>: FreeCAD exception thrown (Mapped element not found: ;#24c1;:M;CUT;:T19:5:F.Face21)
```

...and then the crash happened.

# Additional comment 

This kind of exceptions ("Mapped element not found") had been happening for quite some time (like for a week or two) but I couldn't find a clue about its reason. 

# Backtrace
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7fa6b1df20c0]
#1  0x7fa6b5dbae70 in Gui::Document::getViewProvider(App::DocumentObject const*) const from /opt/FreeCAD/lib/libFreeCADGui.so+0
#2  0x7fa624445407 in TechDrawGui::QGIView::drawCaption() from /opt/FreeCAD/lib/TechDrawGui.so+0x157
#3  0x7fa62444553b in TechDrawGui::QGIView::drawBorder() from /opt/FreeCAD/lib/TechDrawGui.so+0x1b
#4  0x7fa624443b09 in TechDrawGui::QGIView::hoverEnterEvent(QGraphicsSceneHoverEvent*) from /opt/FreeCAD/lib/TechDrawGui.so+0x39
#5  0x7fa6aec2aff7 in QGraphicsItem::sceneEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x297
#6  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7e6ba1) [0x7fa6aec5aba1]
#7  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7e6f2f) [0x7fa6aec5af2f]
#8  0x7fa6aec5b47a in QGraphicsScene::mouseMoveEvent(QGraphicsSceneMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8a
#9  0x7fa6aec68f47 in QGraphicsScene::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x137
#10  0x7fa6ae63854c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#11  0x7fa6ae63f52c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#12  0x7fa6b5e0b171 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#13  0x7fa6adb71f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#14  0x7fa6aec7f322 in QGraphicsViewPrivate::mouseMoveEventHandler(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a2
#15  0x7fa6aec7f85b in QGraphicsView::scrollContentsBy(int, int) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1eb
#16  0x7fa6adb86660 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#17  0x7fa6aecd815e in QAbstractSlider::valueChanged(int) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3e
#18  0x7fa6aec794f8 in QGraphicsView::centerOn(QPointF const&) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x188
#19  0x7fa6aec79b89 in QGraphicsViewPrivate::centerView(QGraphicsView::ViewportAnchor) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x69
#20  0x7fa6aec838cb in QGraphicsView::resizeEvent(QResizeEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x7b
#21  0x7fa6ae68ee80 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x290
#22  0x7fa6aea53c8e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e
#23  0x7fa6aec81b74 in QGraphicsView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x184
#24  0x7fa6adb720a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#25  0x7fa6ae63852c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#26  0x7fa6ae63f52c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#27  0x7fa6b5e0b171 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#28  0x7fa6adb71f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#29  0x7fa6ae68b001 in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1f1
#30  0x7fa6ae68c183 in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x33
#31  0x7fa6ae68c0c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x112
#32  0x7fa6ae68c19f in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x4f
#33  0x7fa6ae68c0c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x112
#34  0x7fa6ae68c19f in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x4f
#35  0x7fa6ae68c0c2 in QWidgetPrivate::showChildren(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x112
#36  0x7fa6ae68c19f in QWidgetPrivate::show_helper() from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x4f
#37  0x7fa6ae68e1fa in QWidget::setVisible(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x20a
#38  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x61730f) [0x7fa6aea8b30f]
#39  0x7fa6aea8bcea in QMdiSubWindow::changeEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x19a
#40  0x7fa6ae68ed7b in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x18b
#41  0x7fa6aea8c68b in QMdiSubWindow::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8b
#42  0x7fa6ae63854c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#43  0x7fa6ae63f52c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
[... 53 lines cut ...]
#97  freecad-git(main+0x862) [0x55bbcdb2c7b2]
#98  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7fa6ac42c2e1]
#99  freecad-git(_start+0x2a) [0x55bbcdb2da0a]
FreeCAD 0.18, Libs: 0.18R16566 (Git)
```

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16566 (Git)
Build type: Release
Branch: LinkStage3
Hash: 6b3f0150ae850f58056f43a90bd45a6cda5d471a
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

I'm handing over the model privately.

realthunder, 2019-04-28:
Please sync my branch and try again. See if it happen again.

ceremcem, 2019-04-28:
I'm testing.

ceremcem, 2019-05-01:
There is no crash till this point. I'm going to close as it's not reproducible.

closed 2019-05-01 by ceremcem