realthunder/FreeCAD_assembly3 #243: Issue with subshape binder

ceremcem, 2019-04-28:
# Live Logs 

> This is the same file I've sent you lastly.

1. Suddenly the following part's TechDraw view is created erroneously, as follows: 

[image]

2. I checked the source Fusion:

[image]

3. Considering the dimensions, I found that the `Binder{12,13,14}` seem to be bound to the wrong part (bound to `keystone.asm_cut` instead of `keystone.metal`)

I don't know how this switch has happened. 

# Resolution 

### Failed 1
1. Binder12 -> Synchronize didn't solve the issue 

### Failed 2
1. Selecting both Binder12 and `housing.Parts.back_window.keystone005.Parts.metal` and clicking to "SubShape Binder" button didn't solve the issue either.

### Failed 3
In "driver-base" file, 
1. Selected `housing.Parts.back_window.keystone005.Parts.metal`
2. Created a sub-shape binder (Binder051 is created)
3. Visually verified that newly created binder has the correct dimensions:
[image]
4. Dragged Binder051 into "processed_parts.back_window_processed...Cut005.Fuse"
5. Deleted Binder12
6. Nothing changed, Fusion suddenly has the wrong size. 

### Success 

1. Revert the file to its original (erroneous) state. 
2. See the fusion is wrong. 
3. Navigate to `housing.Parts.back_window.keystone005.Parts` and set `assembly_cut` as invisible.
4. Recompute

> Note: Seems highly related to #242 

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16594 (Git)
Build type: Release
Branch: LinkStage3
Hash: 7b1015190f7086cd6c5d5b52258ce216f8611282
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-04-30:
> Considering the dimensions, I found that the `Binder{12,13,14}` seem to be bound to the wrong part (bound to `keystone.assembly_cut` instead of `keystone.metal`)

In case you didn't know, you can find out the bound object by right click the binder item in tree view, and choose `Select bound objects`.

> 3\. Navigate to `housing.Parts.back_window.keystone005.Parts` and set `assembly_cut` as invisible.

As mentioned in my other [reply](../issues/242#issuecomment-487822872), currently this is a designed behavior. You can try another [feature](../wiki/How-to-Handle-Large-Scale-Assembly#derived-assembly) to have alternative visibility configuration. And yes, your [scene feature](#186) suggestion is on my todo list.

ceremcem, 2019-04-30:
> (from #242) The assembly shape is determined by its parts visibility.

The binders I'm using in boolean operators aren't bound to the **assemblies**, but they are bound to the **parts in assemblies**. 

Sub-shape binder should reflect the exact shape of geometry that it is bound to, not its target geometry's parent assembly container's whole shape, right?

> You can try another feature to have alternative visibility configuration

I haven't used the "freeze" feature until now, it's a good reason for learning about it. This may also help making the project file archive ready (I mean, the project file should be readonly starting from this, stable state. It shouldn't produce any different fabrication output in the future). 

> And yes, your scene feature suggestion is on my todo list.

You guessed right, I would refer to the scene feature :) However, I think the current behavior is confusing, so it would be still confusing to overcome this issue by using scenes (I mean, if scenes were implemented, we had to switch to "Production scene" before recomputing the fabrication outputs.)

realthunder, 2019-04-30:
> The binders I'm using in boolean operators aren't bound to the _assemblies_, but they are bound to the _sub-parts in assemblies_.

But the Binder{12,13,14} are bound to keystone{4,5,6}, which are in fact assemblies, and you are changing the visibility of parts inside those assembly, no?

ceremcem, 2019-04-30:
Well, I think we found the problem. I never meant to bind those binders to the "keystone"s, I bound them to "keystone.metal"s accordingly.

ceremcem, 2019-04-30:
See `# Failed 3`

realthunder, 2019-04-30:
> 1. Binder12 -> Synchronize didn't solve the issue

Before using `Synchronize`, you should verify if the binder is bound to the correct object, by using the `Select bound objects` tree menu action. If it is the wrong object, it won't work.

> 1. Selecting both Binder12 and `housing.Parts.back_window.keystone005.Parts.metal` and clicking to "SubShape Binder" button didn't solve the issue either.

Doing this will not reset binding, but instead add metal as an extra binding. This is kind of dumb, it should be smart enough to remove parent if a child is added to a binding. I'll change that. Anyway to reset binding, drag the bound object on to the binder, but make sure you hold CTRL key while dropping, otherwise, it will add an extra binding as well.

> 1. Selected `housing.Parts.back_window.keystone005.Parts.metal`
> 2. Created a sub-shape binder (Binder051 is created)
> 3. Visually verified that newly created binder has the correct dimensions:
> 4. [...]

It works for me, although it will be easier just drop the on the binder to modify it.

BTW, why are you using App::Part, its requirement on its children is conflicting with my container, you'll get warning by dragging object across App::Part boundary. But I don't think that's the cause of the wrong binding object. Still can't imagine how did that happen.

[image]

ceremcem, 2019-04-30:
> BTW, why are you using App::Part

It was a workaround attempt to [my `.Label` hack for SheetMetal](https://github.com/shaise/FreeCAD_SheetMetal/pull/70), where I'm storing information within the Label. The fragile side is that if you add more features, the object with information in its label is left down in the hierarchy, thus breaking the "hack". I thought I could use the App::Part to keep the processed labeled item (the App::Part) every time in the root. In short, it was a part of my other hack attempt. 

>  it should be smart enough to remove parent if a child is added to a binding. I'll change that.

I appreciate that. 

> Anyway to reset binding, drag the bound object on to the binder, but make sure you hold CTRL key while dropping, otherwise, it will add an extra binding as well.

I was not aware of this option! `T,D`, `go to target`, `Hold Ctrl`, `Click` works perfectly. I'm adding this to [my freecad-notes](https://github.com/ceremcem/freecad-notes#by-task-asm3).

> It works for me,

It turns out I made a mistake at the end of the procedure. So yes, it works for me too.

closed 2019-04-30 by ceremcem