realthunder/FreeCAD_assembly3 #246: Bad Cast Error if Pad Length is Read From a Spreadsheet Cell

USPR, 2019-06-08:


USPR, 2019-06-08:
Steps to reproduce:
1. Create a Spreadsheet, enter a numeric cell value, and give the cell an alias like "testValue".
2. Create a Part Design Body with a Sketch.
3. Create a Part Design Pad.
4. For the Length, enter a formula like "Spreadsheet.testValue".
5. Click OK to finish the Pad dialog.
An "Input Error" message box appears with the text "bad cast".
The pad is not created.

Report view: 
882.485 <App> Document.cpp(3407): exception in Unnamed#Pad thrown: bad cast
882.518 <App> Document.cpp(3407): exception in Unnamed#Pad thrown: bad cast

Appears in Preview 0.10, 2019.06.08.
Worked fine in Preview 0.9.1, 2019.02.02

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16830 +1493 (Git)
Build type: Release
Branch: LinkStage3
Hash: 4f7663413607b93382291316ae61c2c8ef36646f
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

USPR, 2019-06-08:
Sorry for any mistake, this is my first post here. 
Ulrich

realthunder, 2019-06-08:
Thinks for reporting. But bug is confirmed. I'll accumulate the bugs and make a bug fix release in a week or two. Meanwhile, you can work around it by explicitly specify a unit in spreadsheet, e.g. `5mm`, and it will work.

USPR, 2019-06-09:
Great! I'll do that can keep on testing. By the way, the asm3-fork is a very promising approach when it comes to assembly. Thank you very much for your effort. Hopefully, I can contribute, first with tutorials, maybe later with coding.

realthunder, 2019-06-09:
Thanks for the appreciation!

closed 2019-06-09 by realthunder