realthunder/FreeCAD_assembly3 #247: module 'freecad.asm3.py3_slvs._slvs' has no attribute 'System_swiginit'

Marsu31, 2019-06-09:
HI !

When I run the [first tutorial](https://github.com/realthunder/FreeCAD_assembly3/wiki/Common-Operations), the step "Finally, click [Solve icon] to solve the constraint system." produces the error : `module 'freecad.asm3.py3_slvs._slvs' has no attribute 'System_swiginit'`.

The error occurs only with FreeCAD-asm3-Bionic-Py3n2-Qt5-20190608.glibc2.25.

Here are the tested releases :
**KO**
OS: Linux Mint 19.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16830 +1493 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 4f7663413607b93382291316ae61c2c8ef36646f
Python version: 3.6.7
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

**OK**
OS: Linux Mint 19.1 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18320 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 1488153018418e0d8b7357c4a418235231e4c40e
Python version: **3.7.3**  (why ? `python3 -V` gives `Python 3.6.7`)
Qt version: 5.6.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

**OK**
OS: Linux Mint 19.1 Tessa
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16268 (Git) AppImage
Build type: None
Branch: (HEAD detached at FETCH_HEAD)
Hash: 9a4632f4384c8f7496baf75aa38a46be74649d83
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

Regards

realthunder, 2019-06-10:
The `Bionic` image is update. Please try again.

Regarding the Python 3.7.3 thing, are you using the Conda image for that? AppImage packs its own version of python binary inside.

Marsu31, 2019-06-10:
It works !

Python 3.7.3 was Conda image indeed.

Thank you. ^^

closed 2019-06-10 by Marsu31