realthunder/FreeCAD_assembly3 #249: AppImage does not include harfbuzz

bgamari, 2019-06-12:
Attempting to run the Cosmic AppImage on NixOS fails with a dynamic linker error:
```
$ nix run -f . freecad-realthunder-0_10 -c FreeCAD
freecad-python3: error while loading shared libraries: libharfbuzz.so.0: cannot open shared object file: No such file or directory
```
My understanding is that AppImages should be self-contained and the same approach works with the 0.9.10 image so I suspect `libharfbuzz` is missing from the packaging.

realthunder, 2019-06-13:
`libharfbuzz.so.0` is on AppImages (the builder tool) exclude list, which means it is probably a very system aware (i.e. low level) library. You'll have to install it on your system first.

realthunder, 2019-06-13:
Another thing, I am not familiar with Nix, but it seems AppImage have some [trouble](https://github.com/AppImage/AppImageKit/issues/472) on NixOS. Looks like you need a special Nix package [appimage-run](https://github.com/NixOS/nixpkgs/blob/master/pkgs/tools/package-management/appimage-run/default.nix) to run it.

bgamari, 2019-06-28:
Indeed I was using `appimage-run` but apparently too old a version. Using a newer `nixpkgs` fixed this. Thanks!

closed 2019-06-28 by bgamari