realthunder/FreeCAD_assembly3 #254: Alias Manager Macro broken

jonascant, 2019-07-10:
Hi, with this version of FreeCad, the Alias Manager Makro does not work anymore (Window just closes, no change in the Spreadsheat)
The python console only outputs the following:
`>>> Gui.runCommand('Std_DlgMacroExecute',0)`
I saw that the Contents Field adds a " ' " before every name. like " Name " becomes " 'Name ".
Is there a way i can debug this? I have no experience in this field of programming.

realthunder, 2019-07-10:
I have submit a [PR](https://github.com/pgilfernandez/FreeCAD_AliasManager/pull/3) for this problem. You can manually fix it by replacing the Alias_Manager.FCMacro file in your macro directory with [this](https://github.com/realthunder/FreeCAD_AliasManager/blob/master/aliasManager_popup.py).

jonascant, 2019-07-11:
Thank you very much for this!

closed 2019-07-11 by jonascant