realthunder/FreeCAD_assembly3 #255: slvs build error parameter of type 'SolveSpace::Entity *'

mtanalogue, 2019-07-10:
Hi, 

I'm trying to build the solvespace submodule. If I build just the bindings it builds fine, when attempting a complete build I get the following error: 

```
/opt/Assembly3/Ext/freecad/asm3/slvs/src/lib.cpp:158:31: error: cannot initialize a parameter of type
      'SolveSpace::Entity *' with an rvalue of type 'SolveSpace::EntityBase *'
                SK.entity.Add(&e);
                              ^~
/opt/Assembly3/Ext/freecad/asm3/slvs/src/dsc.h:264:17: note: passing argument to parameter 't' here
    void Add(T *t) {
                ^

```

```
/opt/Assembly3/Ext/freecad/asm3/slvs/src/lib.cpp:179:23: error: cannot initialize a parameter of type
      'SolveSpace::Entity *' with an rvalue of type 'SolveSpace::EntityBase *'
        SK.entity.Add(&e);
                      ^~
/opt/Assembly3/Ext/freecad/asm3/slvs/src/dsc.h:264:17: note: passing argument to parameter 't' here
    void Add(T *t) {

```

```
/opt/Assembly3/Ext/freecad/asm3/slvs/src/lib.cpp:241:27: error: cannot initialize a parameter of type
      'SolveSpace::Constraint *' with an rvalue of type 'SolveSpace::ConstraintBase *'
        SK.constraint.Add(&c);
                          ^~
/opt/Assembly3/Ext/freecad/asm3/slvs/src/dsc.h:264:17: note: passing argument to parameter 't' here
    void Add(T *t) {
                ^

```

I'm on Ubuntu 18.04, building with clang-8, (it fails with GCC too)

Any suggestions?

closed 2019-07-11 by realthunder

realthunder, 2019-07-11:
Fixed. Please sync asm3 repo with `git pull && git submodule update`. You'll need to redo cmake configure for solvespace. BTW, I am actually using an older solvespace branch. You probably want the upstream if you want to try the full solvespace CAD program.

mtanalogue, 2019-07-11:
thanks. Awesome work by the way.

commit 99f58118 (2019-07-11): Update slvs submodule

Fixes #255