realthunder/FreeCAD_assembly3 #256: Why doesn't this "PointInPlane" constraint work?

ceremcem, 2019-07-30:
I'm pretty sure that I did the very same thing million times:

[image]

However, this time it throws the "Inconsistent Constraints" error. 

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18668 (Git)
Build type: Release
Branch: LinkStage3
Hash: c45b815a3b34f9a690753c6d3db91c13baaeb4d2
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

[fem.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3445655/fem.FCStd.zip)

ceremcem, 2019-07-30:
Likewise, those two "MultiParallel" constraints are unexpectedly conflicting: 

[image]

# Attachments 

[fem.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3445793/fem.FCStd.zip)

realthunder, 2019-07-30:
I think the problem is the `AxialConstraint` and the following `PointsCoincident`, which causes over constraint. The solver can handle them in some cases, but may fail if additional constraints are added. Why do you need these two instead of a single `PlaneCoincident`?

ceremcem, 2019-07-30:
> Why do you need these two instead of a single `PlaneCoincident`?

Not something for a particular purpose, I just forgot that we have `PlaneCoincident` and used those two  constraints to build the `PlaneCoincident` on my own. 

Should we post process our model by searching for such constraints which can be "reduced" by replacing with a single constraint? In other words, is it a **required** approach or an optional approach to reduce the number of constraints by using dedicated/integrated constraints in place of discrete constraints?

realthunder, 2019-07-30:
When you add the `PointsConcident` after the `AxialConstraint`, you will start seeing the `redundant constraint` warning. This is always a bad sign. If you build your assembly by constrain yours parts in a chain, then it's probably fine. But in many cases, there will be loops, and that's when the solver starts to fail. If you developed the habit of watching for that warning, then you can catch the problematic constraint earlier. If you missed it, and got `inconsistent constraint` error later, you can disable the constraint one by one from the bottom up to find out which one is causing the problem.

I'll see if I can make this automatic. I guess it will be straightforward if there is only one redundant constraint, but no so if there are more.