realthunder/FreeCAD_assembly3 #257: Automatically open required view on "Select bound object"

ceremcem, 2019-08-06:
"Select bound object" needs the bound object's view to be opened beforehand. The context menu entry should make the view open for instead of displaying that error: 

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18668 (Git)
Build type: Release
Branch: LinkStage3
Hash: c45b815a3b34f9a690753c6d3db91c13baaeb4d2
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

closed 2019-08-06 by realthunder

realthunder, 2019-08-06:
Should be fixed now. BTW, If you sync with my branch now, you'll get the commit that enables saving to and restoring directly from directory, for use with git. You are welcome to try it. However, I don't consider it stable yet. Please backup with ordinary archive format.

As I remembered that you are manually unpacking FCStd file for git control, you can now directly open that directory. But when you save it, the files inside your repository will be changed dramatically. There is no interaction with git in FC yet. You'll have to manually add new files to git repo. Unused files will be auto removed from the filesystem by FC, but you'll still have to manually commit the changes using git.

ceremcem, 2019-08-06:
> If you sync with my branch now, you'll get the commit that enables saving to and restoring directly from directory, for use with git. You are welcome to try it.

Horray! We are heading to a versionable project format. That's great news.  

> As I remembered that you are manually unpacking FCStd file for git control

Partially true. I tried to unzip FCStd file manually and add the files to git repo. However, some files were binary and couldn't be stored inside git repo efficiently. Anyway, I'm sure this addition will be stable soon, so let's see where it leads us.