realthunder/FreeCAD_assembly3 #258: Unrelated topological naming exception

ceremcem, 2019-08-06:
In my model, there are no errors present. Here is the reproduction steps: 

* Open the file "button-driver.fcstd"
* Open "driver-base" in the tree view 
* Click "Refresh" to see there are no errors. 
* Open "params" Spreadsheet. 
* Add string "paint_thickness" to a cell (A22)
* wall.Bend002 throws the exception: 

[image]

# Attachments 

(sent privately via FC forum DM)

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18668 (Git)
Build type: Release
Branch: LinkStage3
Hash: c45b815a3b34f9a690753c6d3db91c13baaeb4d2
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-08-06:
I think I'll need to patch Sheet Metal in order to fix this. Python code doeing boolean operation can automatically benefit from my topo naming, but not so when they are generating new geometry, especially in the way that Sheet Metal is doing.

ceremcem, 2019-08-06:
I understand. I must denote that this is - kind of - blocker at the moment.

ceremcem, 2019-08-07:
Update: It's not blocker but high prio, I found a workaround.

realthunder, 2019-08-07:
Did you authored this bent using some older OCC, presumably 7.2? It is a known problem that geometry created by OCC older than 7.3 can experience unexpected topo change when recompute under 7.3.

ceremcem, 2019-08-07:
> Did you authored this bent using some older OCC, presumably 7.2?

That's a high probability. Can't we fix it automatically?

realthunder, 2019-08-07:
If the geometry is authored in 7.2 and has proper topo names, then in most cases it can auto correct itself when recompute in 7.3. Unfortunately, SheetMetal does not have proper topo names yet. I'll see what I can do.

luzpaz, 2021-02-20:
@ceremcem is this still an issue?

ceremcem, 2021-02-20:
@luzpaz Thanks for reminding. This issue is very well solved.

closed 2021-02-20 by ceremcem