realthunder/FreeCAD_assembly3 #259: Temporarily freezing the model except for some assemblies - Design Time Test Case

ceremcem, 2019-08-06:
Let's assume we modelled a fighter plane and we produced a prototype. We produced molds and then parts. Thousands of parts are post processed (bent, drilled, threaded and so on). 

Then we saw by the flight tests that the radar has to be changed with a slightly different model so we have to make the nose con a little bit longer.

We might perfectly have the plane parts assembled relative to the nose cone (this is the scenario) by the assembly constraints, so every drill, every length might be changed by a single cone length change. So our change will cause throwing away all the molds, produced parts etc. 

There should be a mechanism that will warn us. Something like that: 

1. Freeze whole model. 
2. Unfreeze nose cone and its drill base
3. Change the length 
4. FC throws a DTTC (Design Time Test Case) exception: "Your changes causes X change on Y part, Z change on T part, etc."
5. Developer decides that "We have to accept the change on Y part too, but not on T part"
6. Unfreeze part Y. 
7. Make some more changes on the nose cone 
8. FC throws DTTC exception: "Your changes causes K change on M part"
9. Developer makes some more optimizations. 
10. All DTTC exceptions are cleared. 
11. Team reproduces the nose cone and part Y and everything related to those parts. 

The example might have seemed fictional but it's what I've came across at the moment.

realthunder, 2019-08-07:
That wouldn't be a reliable method to check for dependency, because there may be multiple dependency between two parts. Trail and error may missed some give false impression to the user.

There is a "show dependency graph" command, which does not quite work in my branch. I have plan to bring it up to date, just haven't got time to do it yet.

ceremcem, 2019-08-07:
"Dependency graph" sounds quite logical yet a more efficient approach. I understand what you mean by "trial and error approach annoyance". Thinking again, if a change will trigger 1000 errors eventually, this will absolutely make the user upset. On the other hand, a dependency graph would dump all dependencies and dependents; we can predict all errors at once.