realthunder/FreeCAD_assembly3 #260: FreeCAD Crash

ceremcem, 2019-08-07:
# Description 

I selected a part in the TechDraw page, clicked to change its source, deselect it via holding `Ctrl` key and clicking and accepted my changes (so I expected to remove the source); then FC is crashed. 
 
# Backtrace 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7f4442d850c0]
#1  0x7f443eb2b12c in QVariant::toStringList() const from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1c
#2  0x7f44470ae18e in Gui::PropertyEditor::LinkListLabel::onEditClicked() from /opt/FreeCAD/lib/libFreeCADGui.so+0x4e
#3  0x7f443eb19660 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#4  0x7f443fc6ac72 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x42
#5  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59a5c3) [0x7f443f9a15c3]
#6  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59b944) [0x7f443f9a2944]
#7  0x7f443f9a2a54 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x74
#8  0x7f443f621e80 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x290
#9  0x7f443f5cb54c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#10  0x7f443f5d3ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#11  0x7f4446e53251 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#12  0x7f443eb04f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#13  0x7f443f5d1ccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#14  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f443f64d1a9]
#15  0x7f443f64bb5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#16  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7f443f675502]
#17  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f443609a7f7]
#18  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f443609aa60]
#19  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f443609ab0c]
#20  0x7f443eb35854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#21  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f443f6755d6]
#22  0x7f443eb037ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#23  0x7f443eb03b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#24  0x7f443eb09bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#25  0x7f4446dc6476 in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x19c6
#26  freecad-git(main+0x8aa) [0x5577b186384a]
#27  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f443d6ff2e1]
#28  freecad-git(_start+0x2a) [0x5577b186499a]
FreeCAD 0.19, Libs: 0.19R18682 (Git)
```

(Note that recovery is failed without a reasonable explanation)

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18682 (Git)
Build type: Release
Branch: LinkStage3
Hash: bf3031326ef4fa8c14e83bca3b0470680e58f8f4
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-08-07:
Can you please go to `/tmp` directory and find any directory with name starting with `FreeCAD_Doc_`, that has a timestamp close to the crash time? Next time if a recovery failed,  please go there and preserve the directory and send to me. Thanks.

closed 2019-08-07 by realthunder

ceremcem, 2019-08-07:
Sure I can send them. Are they sensitive files that I should hand over privately?

realthunder, 2019-08-07:
Yes, they are essential a autosaved copy of your modified document. One directory corresponds to one document. So in your case, there may be multiple directories in `/tmp`

ceremcem, 2019-08-07:
Currently my `/tmp` directory contains the following folders: 

```
drwxr-xr-x 1 aea aea  84 Aug  6 22:54 FreeCAD_Doc_2d8f8b3d-dee2-4c96-83d0-3ac1317861a8_64ef72_13243
drwxr-xr-x 1 aea aea 132 Aug  6 22:54 FreeCAD_Doc_35ec02bb-b0fb-474c-9005-3144f35541da_275b31_13243
drwxr-xr-x 1 aea aea 196 Aug  6 22:54 FreeCAD_Doc_9546b577-d52c-4bfb-b9fa-d23afdfef497_d5ccd0_13243
drwxr-xr-x 1 aea aea 196 Aug  6 21:05 FreeCAD_Doc_9546b577-d52c-4bfb-b9fa-d23afdfef497_d5ccd0_1668
drwxr-xr-x 1 aea aea 134 Aug  6 22:54 FreeCAD_Doc_971925c9-404b-4802-a4df-63118b2d2745_928861_13243
drwxr-xr-x 1 aea aea 134 Aug  6 22:54 FreeCAD_Doc_971925c9-404b-4802-a4df-63118b2d2745_de0b4a_13243
drwxr-xr-x 1 aea aea 120 Aug  6 22:54 FreeCAD_Doc_a45ac8fb-1f9b-4848-b5dd-e97bc76ffe9c_b8a0d5_13243
drwxr-xr-x 1 aea aea  84 Aug  6 22:54 FreeCAD_Doc_aa6bc5a7-c39d-4349-beec-8c36b2c0464b_7d35fd_13243
drwxr-xr-x 1 aea aea  84 Aug  6 22:54 FreeCAD_Doc_bea2ac8d-415b-48ec-9ef9-0f3648f44c75_2940d2_13243
drwxr-xr-x 1 aea aea  84 Aug  6 22:54 FreeCAD_Doc_bea2ac8d-415b-48ec-9ef9-0f3648f44c75_be161a_13243
drwxr-xr-x 1 aea aea  84 Aug  6 22:54 FreeCAD_Doc_bea2ac8d-415b-48ec-9ef9-0f3648f44c75_ece7af_13243
```

Do you want me to send them or we need those in the next time that a recovery is failed?

realthunder, 2019-08-07:
Next time it fails. Do not close the recovery window. Just go there and pack all the directories first.

ceremcem, 2019-08-07:
Got it.

ceremcem, 2019-08-08:
I forcefully closed FreeCAD while it was busy with opening my project. There were lots of errors being logged in the Report View and lots of CPU was being used, which I think due to trying to open a project with a slightly older version of FC. I packed all "FreeCAD_Doc*" folders while the following window is open: 

[image]

# Attachments 

(sent via FC forum DM)

realthunder, 2019-08-09:
There is a related fix with this problem, and more. Please sync my branch.