realthunder/FreeCAD_assembly3 #261: Error: Reading failed from embedded file

ceremcem, 2019-08-08:
I'm seeing this error when I first open my project: 

```
Reading failed from embedded file: GuiDocument.xml (103937 bytes, 4204 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (384668 bytes, 14362 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (192063 bytes, 7496 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (289630 bytes, 11173 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (192063 bytes, 7321 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (158277 bytes, 5993 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (212323 bytes, 8154 bytes compressed)
Reading failed from embedded file: GuiDocument.xml (304549 bytes, 11652 bytes compressed)
```

It doesn't seem to do anything harmful, but they are printed in red.

realthunder, 2019-08-08:
Can you please send me the files that caused these errors?

ceremcem, 2019-08-08:
Link sent via FC forum DM.

closed 2019-08-08 by realthunder

ceremcem, 2019-08-08:
I'm still experiencing those errors: 

```
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
<Exception> End of document reached
<Base> Reader.cpp(710): Reading failed from embedded file: GuiDocument.xml
Rebuilding Views for: Page/pcb-layout
Rebuilding Views for: Page/2mm_metal
Rebuilding Views for: Page001/control-pcb001
DVP::execute - No Sources (but document is restoring) - ProjItem015
Rebuilding Views for: Page002/2mm_plexi
DVP::execute - No Sources (but document is restoring) - ProjItem015
Rebuilding Views for: Page/2mm_metal
```

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18686 (Git)
Build type: Release
Branch: LinkStage3
Hash: d3a1d324d449c1b30720a7d16a325a046e58c86c
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2019-08-08:
You'll need to manually save all the files once to correct the problem. The error is caused by not saving the camera setting because those files don't have views open.

ceremcem, 2019-08-08:
I thought I already did that. Should I do something different: 

* I opened the project file 
* Errors are produced 
* Clicked "Close"
* File save dialog appeared
* Answered: "Save all"
* Reopened the same project file 
* Errors are produced
* Clicked some 3D model in the tree view
* Clicked to close the file 
* File save dialog appeared
* Answered: "Save all"
* Reopened the same project file 
* Errors are produced

Should I open every 3D model separately and then save?

realthunder, 2019-08-08:
`Save all` only saves the modified files. No need to open them separately. Just open the top level document as normal. For each auto opened files, click any 3D object inside to make sure there is a corresponding tab of that document opened. Then click `Save` to manually save that document.

ceremcem, 2019-08-08:
Ok, now the error is cleared. Thanks.