realthunder/FreeCAD_assembly3 #262: Error: Sub-object not found

ceremcem, 2019-08-08:
# Description 

When I run [this macro](https://github.com/ceremcem/FreeCAD_SheetMetal/blob/7b401415c47c5e9f86b31ead0153743f0559b85a/Macros/SheetMetalUnfoldUpdater.FCMacro) Report View outputs the following error: 

```
17581.2 Selection.cpp(1558): Sub-object driver_base#Assembly.Parts.Group001.Link003.Parts001.Fusion006.Face29 not found
```

However I can't see any exclamation mark in front of any objects. It reminds me the errors when a geometry is dramatically changed and a face or edge can not be found because of highly changed topology. However, I can't track the erroneous part down: 

[image]

# Attachments 

The same files #261 

# Version 

Same as #261

realthunder, 2019-08-08:
The error is emitted by Gui.Selection, because the selected object/element can no longer be found, for example when some object is removed. I should probably downgrade it to a warning. To avoid the error, clear the selection before removing object or making geometry changes in the macro.

ceremcem, 2019-08-08:
I had a few unsuccessful tries. What is the best way to add a face to selection that an `Element` points to? My current approach is as follows: 

```py
my_elem_label = "foobar"
face_elem = doc.getObjectsByLabel(my_elem_label)[0]
asm_container = face_elem.Parents[0][0]
target_obj_path = asm_container.getSubObjects()[0]
unfold_face_id = (asm_container.Name, target_obj_path + face_elem.LinkedObject[1].split('.')[-1]) 
			
Gui.Selection.clearSelection()
Gui.Selection.addSelection(doc.Name, unfold_face_id[0], unfold_face_id[1])
```

realthunder, 2019-08-09:
An `Element` may point to some other intermediate elements. To select what it points to directly,

```python
element = App.ActiveDocument.Element001.Proxy
Gui.Selection.addSelection(element.getAssembly().getPartGroup(), element.getElementSubname())
```
If you want to skip intermediate elements, than add a `True` argument to `getElementSubname()` call.

There is a trick for more advanced operations in Macro. Observe the Python console when you click those toolbar buttons. Say, you select an element, and click `Select linked object` button. You will see the Python console outputs the corresponding command.

```python
Gui.runCommand('asm3CmdGotoLinked',0)
```
That's how Macro recording works actually. All these commands uses the current selection as inputs and perform some actions. You can try this with other commands.

realthunder, 2019-08-09:
There are other commands that output partial Python code into the console, which are not helpful for Macro. Those command will be appended with a comment like

```python
### Begin command asm3CmdGotoLinkedFinal
```

And this is how you can find out the command name. So you can invoke the command by

```python
Gui.runCommand('asm3CmdGotoLinkedFinal')
```

ceremcem, 2019-08-09:
The 

```
Gui.Selection.addSelection(element.getAssembly().getPartGroup(), element.getElementSubname())
```

has solved the problem,  I'm not getting that error any more. Thanks. This also closes #236, yes?

closed 2019-08-09 by ceremcem