realthunder/FreeCAD_assembly3 #264: Erroneous Element Location

ceremcem, 2019-08-08:
Location of bolts and its related constraints' elements are not displayed at the same point: 

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18686 (Git)
Build type: Release
Branch: LinkStage3
Hash: d3a1d324d449c1b30720a7d16a325a046e58c86c
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

Sent via FC forum DM.

realthunder, 2019-08-08:
I didn't see the offset when opening your document. Can you repeat the steps, and capture a screenshot?

ceremcem, 2019-08-08:
I couldn't reproduce the case in the above screenshot. However, replacing the constraints with a different type produced a similar result: https://youtu.be/y6rAuxlzfPY

ceremcem, 2019-08-08:
Currently I can see no errors in the Report View but constraints are not being solved.

ceremcem, 2019-08-08:
In the current project, solver is not working: 

[image]

# Attachments 

Sent via FC forum DM

realthunder, 2019-08-08:
Haven't got time to check the solver problem yet, but the element position problem should be fixed

closed 2019-08-08 by realthunder

ceremcem, 2019-08-08:
It seems that the solver problem is solved too. I'll open a separate issue if anything goes wrong. Thanks!