realthunder/FreeCAD_assembly3 #265: Feature: Disable/Enable Tree View Tracking

ceremcem, 2019-08-09:
When we click a geometry, tree view immediately jumps to the relevant object (or feature). Most of the times this is good. However, if there are lots of (lots of) broken `Element`s, it might be more helpful to be able to disable this feature temporarily. 

# Current flow 

1. Click on the erroneous `Element`
2. There will be a yellow "shadow" appeared somewhere in the model 
3. Make a good guess of relevant geometry
4. **[x]** Memorize the `Element` label (eg. `_Element231`)
5. Click on the geometry in the model 
6. **[x]** Relevant object will be automatically selected in the tree view 
7. **[x]** Scroll up and find your `Element` (`_Element231` in this case)
8. Hover on `_Element231`
9. Type `T, D` to  update element link. 

# Proposed flow 

1. Right click on the tree view and "Disable Tree View Tracking"
2. Click on the erroneous `Element`
3. There will be a yellow "shadow" appeared somewhere in the model 
4. Make a good guess of relevant geometry
5. Click on the geometry in the model 
6. Hover on your `Element` (extra feature: This previous item might be indicated with a yellow shadow)
7. Type `T, D` to  update element link. 

# Rationale 

If you have to update +5 elements, this will gain a huge time.

realthunder, 2019-08-09:
The feature is there. It is called 'Sync selection', along with other tree view actions, including the drag action. If you don't like the default shortcut, you can customize it using `Tools -> Customize -> Keyboard -> TreeView`

[image]

ceremcem, 2019-08-10:
I almost never touched those features (because of the *serious* (and *dangerous* for me) "sync placement" option) since the "sync placement" feature is first introduced. 

Anyhow, "Sync selection" is exactly what I meant. Thanks for the feature from future.

I think a dedicated documentation page (kind of a cheatsheet) only for those options would be very useful. Here is my knowledge so far: 

# Context menu 

* **Sync view**: 
  * Description: Auto switch to the 3D view containing the selected item.
  * Shortcut: `T, 1`
  * When you want to disable: 
    * If a module (eg. TechDraw) wants you to to select an object in the tree view and then press a button (eg. TechDraw's "Insert multiple linked views of drawable object(s)" button), you *might* want to disable. However there is a tip: First select the object you want to "Insert its linked view", then click `Ctrl` key, then click your TechDraw instance. Last selection will switch to TechDraw view, so you can click "Insert multiple linked views..." button without disabling this feature. 

* **Sync selection**:
  * Description: Auto expand tree item when the corresponding object is selected in 3D view.
  * Shortcut: `T, 2`
  * When you want to disable: 
    * While fixing multiple `Element`s: 
      * `1.` Disable *Sync selection*
      * `2.` Click on the erroneous `Element`
      * `3.` There will be a yellow "shadow" appeared somewhere in the model 
      * `4.` Make a good guess of relevant geometry
      * `5.` Click on the geometry in the model 
      * `6.` Hover on your `Element` (extra feature: This previous item might be indicated with a yellow shadow)
      * `7.` Type `T, D` and then "Click" to update element link. 
      * `8.` Go to step 2 until there is no broken `Element` left. 
      * `9.` Enable *Sync selection*

* **Sync placement**:
  * Description: Auto adjust placement on drag and drop objects across coordinate systems.
  * Shortcut: `T, 3`
  * When you want to disable: 
    * **Never?** 
* **Pre-selection**:
  * Description: Pre-select the object in 3D view when mouse over the tree item.
  * Shortcut: `T, 4`
  * How to use: 
    * Hover an object on tree view. 
    * See the object is highlighted with yellow color and brought to the front in the 3D view so you can easily locate the object.
  * When you want to disable: 
    * **Never?** 
* **Record Selection**:
  * Description: Record selection in tree view in order to go back/forward using navigation button.
  * Shortcut: `T, 5`
  * How to use: **???**
  * When you want to disable: 
    * **???**

**Question:** Where is the "navigation button" mentioned in "Record Selection" description? How do we benefit from "Record Selection" feature? What is the use case?

closed 2019-08-10 by ceremcem

realthunder, 2019-08-10:
> **Question:** Where is the "navigation button" mentioned in "Record Selection" description? How do we benefit from "Record Selection" feature? What is the use case?

It is documented [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/Navigation). It will be very handy if combined with other navigation actions, such as `select (deepest) linked object`, especially those long ranged cross document jumps.

ceremcem, 2019-08-10:
Got it.

What do you think my claims for the features those are "Not meant to be disabled"? Is there any case that we might want them to be disabled?

realthunder, 2019-08-10:
`Sync placement` may interfere with some container object if it implements its own placement management.

`Pre-selection` may cause slow down when dealing with very large model. When graphics memory is not enough, the slowdown will be very noticable.

`Record selection` can be temporarily disabled to make the last selection as an anchor to jump back,  while ignoring any follow up selection.

ceremcem, 2019-08-10:
Thanks. I made a clean copy of above cheatsheet: https://github.com/ceremcem/freecad-notes/blob/master/tree-view-actions-cheatsheet.md