realthunder/FreeCAD_assembly3 #267: FreeCAD crash

ceremcem, 2019-08-12:
# Reproduction 

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18706 (Git)
Build type: Release
Branch: LinkStage3
Hash: 2eee64759656925a5b8ec754b41ae4f134e51001
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Traceback 
```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f2466041060]
#1  0x7f246eebc819 in App::TransactionObject::applyChn(App::Document&, App::TransactionalObject*, bool) from /opt/FreeCAD/lib/libFreeCADApp.so+0x759
#2  0x7f246eebb75c in App::Transaction::apply(App::Document&, bool) from /opt/FreeCAD/lib/libFreeCADApp.so+0x10c
#3  0x7f246ed7ddc4 in App::Document::_abortTransaction() from /opt/FreeCAD/lib/libFreeCADApp.so+0x64
#4  0x7f246efa9bd9 in App::Application::closeActiveTransaction(bool, int) from /opt/FreeCAD/lib/libFreeCADApp.so+0x209
#5  0x7f23d80804b3 in TechDrawGui::TaskDlgProjGroup::reject() from /opt/FreeCAD/lib/TechDrawGui.so+0x43
#6  0x7f246f9ed263 in Gui::TaskView::TaskView::reject() from /opt/FreeCAD/lib/libFreeCADGui.so+0x43
#7  0x7f2467448660 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#8  0x7f2467448660 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#9  0x7f2468599c72 in QAbstractButton::clicked(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x42
#10  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59a5c3) [0x7f24682d05c3]
#11  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x59b944) [0x7f24682d1944]
#12  0x7f24682d1a54 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x74
#13  0x7f2467f50e80 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x290
#14  0x7f2467efa54c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#15  0x7f2467f02ca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#16  0x7f246f781a41 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#17  0x7f2467433f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#18  0x7f2467f00ccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#19  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f2467f7c1a9]
#20  0x7f2467f7ab5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#21  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7f2467fa4502]
#22  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f245e9c97f7]
#23  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f245e9c9a60]
#24  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f245e9c9b0c]
#25  0x7f2467464854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#26  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f2467fa45d6]
#27  0x7f24674327ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#28  0x7f2467432b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#29  0x7f2467438bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#30  0x7f246f6f5006 in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x19c6
#31  freecad-git(main+0x8aa) [0x55ff38ae784a]
#32  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f246602e2e1]
#33  freecad-git(_start+0x2a) [0x55ff38ae899a]
FreeCAD 0.19, Libs: 0.19R18706 (Git)
```

realthunder, 2019-08-12:
I can't reproduce it. Which version of file are you using?

ceremcem, 2019-08-12:
I sent the project files by FC forum DM.

closed 2019-08-12 by realthunder

ceremcem, 2019-08-12:
Not crashing anymore. Thanks.