realthunder/FreeCAD_assembly3 #270: 'ViewProviderAsmConstraint' object has no attribute 'ViewObject'

ceremcem, 2019-08-16:
# Reproduction 

[image]

# Attachments and Version 

Same with #269

realthunder, 2019-08-18:
I can't reproduce this. Can you please attach a sample file?

BePeMee, 2022-07-31:
I have a similar error when I try to apply a constraint. 
[image]
In the above example I have created 2 parts in part design workbench. Then I made a 3rd file in which I created the assembly. I have linked the bodies of the parts (As and pasgat) to the assembly file and put them in the "parts" folder of the assembly. I have followed this tutorial (but with other parts): https://www.youtube.com/watch?v=Q6pbU2dVCfQ&ab_channel=JokoEngineeringhelp
Now, when I try to add a constraint I get the error message.
07:56:42  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAzXcduO/usr/Mod/asm3/freecad/asm3/assembly.py", line 333, in claimChildren
    return getattr(self.ViewObject.Object, 'Group', [])
<class 'AttributeError'>: 'ViewProviderAsmConstraint' object has no attribute 'ViewObject'

Some additional information:
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.709.28846 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c880c050c5f7601e4bae6740de71500edf2be0c7
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)

The sample files:
[assembly3test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9228152/assembly3test.zip)