realthunder/FreeCAD_assembly3 #272: How to refer a constraint inside a subassembly?

ceremcem, 2019-08-16:
In the following case: 

[image]

Inserting `=<<MeasurePointPlane>>.Distance` inside a cell refers to `Assembly.Constraints.MeasurePointPlane`. How can we refer to `Assembly001.Constraints.MeasurePointPlane`? 

# Attachments 

[20190816T2241-refer-to-sub-assembly-constraint.zip](https://github.com/realthunder/FreeCAD_assembly3/files/3510925/20190816T2241-refer-to-sub-assembly-constraint.zip)

realthunder, 2019-08-17:
I am not with my working computer, so cannot try right now. But can you try `<<Assembly001>>.<<Constraints.MeasurePointPlane.>>.Distance`. Some explanation. The first level object reference can only be applied to objects within the same document as the spreadsheet. The second level sub-object path can be arbitrary long and across multiple documents. 

Just a reminder, you should be careful with future dependency issue with such long distance reference.

realthunder, 2019-08-17:
Another way to add the reference, assuming you know which document this object belongs to. Just `<<DocumentName>>#<<MeasurePointPlane>>.Distance`. This way of reference has less dependency problem. For the previous way, you are depending on all children of Assembly001 because of the top parent reference.

ceremcem, 2019-08-17:
`<<DocumentName>>#<<MeasurePointPlane>>.Distance` works like a charm. Thanks.

closed 2019-08-17 by ceremcem