realthunder/FreeCAD_assembly3 #273: Solver does not update

ceremcem, 2019-08-16:
# Description 

In the attached model, the following component (`back-window-cover`) should be symmetrical according to the bolt holes: 

[image]

which is guaranteed by of the following constraints: 

```
driver_base#Assembly.Constraints.Constraint103 (PointInPlane005)
driver_base#Assembly.Constraints.Constraint111 (Symmetric)
```

However it isn't symmetrical (notice the different distances): 

[image]

# Findings  

I started to narrow down the problem first by revealing the source of the asymmetry (bolts or the `back-window-cover`) about the plane by measuring the distances. I preferred using Assembly3 measurement tool thus I added a constraint between the following elements: 

[image]

(Would you please allow selection view window contents to be copied to the clipboard?)

...which suddenly caused an unrelated `Element` to fail: 

[image]

```
PropertyLinks> PropertyLinks.cpp(358): driver_base#Element005.LinkedObject missing element reference driver_base#Cut008 ;#404b;:T6707:5:F.Face28
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 599, in execute
    self.getElementSubname())
  File "/opt/FreeCAD/Ext/freecad/asm3/assembly.py", line 1337, in getElementInfo
    '{}.{}'.format(objName(part),subname))
<type 'exceptions.RuntimeError'>: Failed to get geometry element from driver_base#Cut008.;#404b;:T6707:5:F.Face28
596.440224 <asm3.main> assembly.py(3295): solver exception when auto recompute
Traceback (most recent call last):
  File "<string>", line 515, in _catch
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 416, in solve
    return _solve(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 382, in _solve
    'Failed to recompute {}'.format(objName(assembly)))
RuntimeError: Failed to recompute driver_base#Assembly004
```

Note that this last failure seems to be related to #258 (and I think what I'm talking in #264 is exactly this matter).

Fixing the failing element `connector_background_unfold_face` by selecting the unfold face, dragging and dropping onto the element fixes everything: The solver starts solving and the `back-window-cover` becomes symmetric: 

[image]

Issue can be interpreted as a "silent exception" that blocks the solver from continuing further calculations. 

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18707 (Git)
Build type: Release
Branch: LinkStage3
Hash: 5a598e96321da61fcb25e52beca6a33454ac5e40
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

Sent by FC forum DM.

ceremcem, 2019-08-20:
Any diagnosis on this one?

realthunder, 2019-08-20:
Not yet. Got caught with something else. Will check soon.

realthunder, 2019-08-20:
Here is the problem I found. Binder047 is not synchronized with its bound object. 
[image]

The question is when and how did this happen. It's not obvious by looking at the code. Have you accidently caused any dependency loop and manually corrected it afterwards? This is one possible cause of missing update that I can think of, although not sure.

The missing element problem would have shown up earlier if the Binder is in sync, i.e. recomputed earlier. The direct cause is because of SheetMetal has incomplete support of the topo naming feature. I'll add that support for sure. But may have to wait a few days.

ceremcem, 2019-08-20:
> The question is when and how did this happen. 

I'm currently not able to verify your finding by looking at `Support` property, right? 

[image]

If so, I've several backups in the meanwhile, I can send you all of them if it's useful for this case. 

> Have you accidently caused any dependency loop and manually corrected it afterwards?

I don't remember but probably, yes. 

> But may have to wait a few days.

It's okay to wait for a few days. I'll be alerted in the meanwhile if any change is required (though there will be no production planned in this week).

realthunder, 2019-08-20:
> I'm currently not able to verify your finding by looking at `Support` property, right?

No, you can. Just use 'Select bound object', and reveal the hidden bend. You'll find that the two object are slightly different, which means the binder missed some update of the bend in the past. Once you manually fix the element, the file is fine. But then, without a proper fix of the SheetMetal, you'll have to repeat this process more often.

[image]

ceremcem, 2019-08-21:
In my backup named "20190808T0422-seems-working" there is no such sub-shape binder. In the backup named "20190812T1606-related-with-issue-267" the sub-shape binder is present and is at the same wrong spot: 

[image]

ceremcem, 2020-02-11:
Hi Realthunder, 

> The direct cause is because of SheetMetal has incomplete support of the topo naming feature. I'll add that support for sure. But may have to wait a few days.

Do you have any chance to take a look at this one? Using sheetmetal wb became a nightmare because of this one.

realthunder, 2020-02-15:
> Do you have any chance to take a look at this one? Using sheetmetal wb became a nightmare because of this one.

There is some progress now. You are welcome to test it with my [LinkDev](https://github.com/realthunder/FreeCAD/tree/LinkDev) branch. This branch has lots of new stuffs merged recently, and I haven't got time to test them all yet. You'll also need my fork of [SheetMetal](https://github.com/realthunder/FreeCAD_SheetMetal).

The first time you open documents with SheetMetal object, they will be touched for element name update. And this process may (or may not) break existing sheet metal objects. 

[image]

ceremcem, 2020-02-16:
FYI: Currently can not build due to #279.

ceremcem, 2020-02-18:
When I first open the project, there are lots of warnings saying 
```
Warning - no numeric format in formatSpec  - Dimension
```

Ignoring them and doing a refresh throws the following error: 

```
Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 627, in execute
    self.getElementSubname())
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1406, in getElementInfo
    '{}.{}'.format(objName(part),subname))
<type 'exceptions.RuntimeError'>: Failed to get geometry element from driver_base#Body010.Bend003.;#1417;:T5825:5:E.Edge52
Recompute failed! Please check report view.
```

There seems one Element broken. I tried to fix it manually, but it throws the same error:

[image]

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21157 (Git)
Build type: Release
Branch: LinkDev
Hash: e24bc133c32edc7bf09616ad609725dd49474951
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachment

The same project I've sent.

realthunder, 2020-02-26:
Sorry for the delay. Lots of fixes. Please sync both LinkDev and asm3 before trying. I have added a magic button,
[image]

It is a checkable option. Once enabled, it will enable auto element fix on recomputation. And I have tried your file, no element reports error with this option enabled. But you'll still have to check your models to make sure. The auto fix utilize my latest feature to search shape by doing geometry comparison in addition to topo name mapping. It takes advantage of the fact that each element saves a copy of the geometry shape.

This auto fix feature is still experimental. You can disable it, and do manual fix using context menu 'Fix' on each broken element and see if the fix is correct.

ceremcem, 2020-02-26:
I enabled the auto fix feature, made a recompute. There were some things happened, I guess which are good: 

```
<PropShape> PropertyTopoShape.cpp(359): Recomputation required for document 'driver_base' on geo element version change: 1.14.70200.3.0 -> sm1.1.14.70200.3.0
QGISVGTemplate::createClickHandles - xml loading error
57.18282 <asm3.main> assembly.py(676): driver_base#Bend003 reference change driver_base#Group004.Body010.Bend003. Edge52 -> Edge31
<App> PropertyLinks.cpp(358): driver_base#_Element260.LinkedObject missing element reference driver_base#Fusion006 ;#5692;:T6063:5:E.Edge134
<App> PropertyLinks.cpp(358): driver_base#_Element259.LinkedObject missing element reference driver_base#Fusion006 ;#568a;:T6063:5:E.Edge136
<App> PropertyLinks.cpp(358): driver_base#_Element263.LinkedObject missing element reference driver_base#Fusion006 ;#5707;:M#5706;RFI;:T6165:5:F.Face43
<App> PropertyLinks.cpp(358): driver_base#_Element262.LinkedObject missing element reference driver_base#Fusion006 ;#56e5;:M;FUS;:T6063:5:E.Edge12
<App> PropertyLinks.cpp(358): driver_base#_Element255.LinkedObject missing element reference driver_base#Fusion006 ;#56ae;:T6063:5:E.Edge161
<App> PropertyLinks.cpp(358): driver_base#_Element256.LinkedObject missing element reference driver_base#Fusion006 ;#56aa;:T6063:5:E.Edge157
<App> PropertyLinks.cpp(358): driver_base#_Element258.LinkedObject missing element reference driver_base#Fusion006 ;#56b0;:T6063:5:E.Edge163
<App> PropertyLinks.cpp(358): driver_base#_Element257.LinkedObject missing element reference driver_base#Fusion006 ;#56ac;:T6063:5:E.Edge159
0.991099 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge161 -> Edge161
0.011637 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge157 -> Edge157
0.324339 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge159 -> Edge159
0.009902 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge163 -> Edge163
0.010845 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge12 -> Edge12
0.009316 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Face43 -> Face43
0.016961 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge136 -> Edge136
0.028527 <asm3.main> assembly.py(676): driver_base#Fusion006 reference change driver_base#Fusion006. Edge134 -> Edge134
<App> PropertyLinks.cpp(358): button_driver#Element001.LinkedObject missing element reference button_driver#Cut ;#5a8a;:M;CUT;:T19:5:F.Face17
0.313816 <asm3.main> assembly.py(676): button_driver#Cut reference change button_driver#Cut. Face17 -> Face17
<App> PropertyLinks.cpp(358): driver_base#Element009.LinkedObject missing element reference driver_base#Cut005 ;#6286;:T6327:5:F.Face36
0.854825 <asm3.main> assembly.py(676): driver_base#Cut005 reference change driver_base#Cut005. Face36 -> Face36
59.945784 <asm3.sys> sys_slvs.py(65): dof remaining: 2
0.34688 <asm3.sys> sys_slvs.py(60): redundant constraints
0.000111 <asm3.sys> sys_slvs.py(65): dof remaining: 0
0.024897 <asm3.sys> sys_slvs.py(65): dof remaining: 3
```

Now it works nicely. Thanks for the fix!

closed 2020-02-26 by ceremcem