realthunder/FreeCAD_assembly3 #279: error: pathspec 'slvs' did not match any file(s) known to git.

ceremcem, 2019-10-26:
Currently an error is thrown after [the following command](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction#solvespace): 

```cmd
$ cd asm3
$ git submodule update --init slvs
error: pathspec 'slvs' did not match any file(s) known to git.
```

It seems that the submodule has been removed after [this commit](https://github.com/realthunder/FreeCAD_assembly3/blob/dcef039b6f31dc8c400b3bbe74ab7692087751f1/.gitmodules#L1). How can we install `slvs` in current state?

realthunder, 2019-10-28:
I am preparing py_slvs as a pip installable [package](https://github.com/realthunder/slvs_py). I only got time to test building linux wheel, no windows or mac, therefore I haven't submit the package to PyPi yet. For now, you can manually clone [py_slvs](https://github.com/realthunder/py_slvs) (or [py3_slvs)[https://github.com/realthunder/py3_slvs) if you are using Python3) under `<your_repo_dir>/freecad3/asm3/`

ceremcem, 2019-10-28:
I'm building against Python2 (I don't know why). However, no need to rush because I rolled back to my previous VM, so currently everything works smoothly.

ceremcem, 2020-02-16:
It's interesting why I didn't correctly follow such an easy workaround at that time. Replacing 
```
git submodule update --init slvs
```
with 
```
rm -r py_slvs 2> /dev/null
git clone https://github.com/realthunder/py_slvs
```

made the build progress. 

I'm still leaving this issue open for anyone following the [wiki/solvespace](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction#solvespace) instructions.

realthunder, 2020-02-16:
FYI, py_slvs PyPi package is online now. You can install it through pip,

[code]
pip install py-slvs
[/code]

ceremcem, 2020-02-26:
I tried to update the Wiki but I think I'll make a mess in the end because I ended up deleting nearly everything related to building `slvs`.

timmwille, 2021-01-23:
> It's interesting why I didn't correctly follow such an easy workaround at that time. Replacing
>
> ```
> git submodule update --init slvs
> ```
>
> with
>
> ```
> rm -r py_slvs 2> /dev/null
> git clone [...]

thx @ceremcem I'm having the same issue and still missing one part of the puzzle 
this here might hold some answers (as my Manjaro works with python >3.9) 

The Arch AUR PKGBUILD is outdated as I figuered:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freecad-assembly3-git

but even with your mentioned fix I could not get it to work
I also tried with `pip install py-slvs` as mentioned by @realthunder but might not work due to the PKGBUILD file (I will edit and test, if you have suggestions)

ceremcem, 2021-01-23:
Python3 instructions is here: https://github.com/realthunder/FreeCAD_assembly3/issues/363#issuecomment-737404792

Does it resolve your case?

timmwille, 2021-01-23:
Wow fast reply :) I'll give it a try, it looks good, just need to find a right code for making the directory generalised:

`asm3=path/to/fc-build/Mod/asm3/freecad/asm3`

is the same as `"${srcdir}/${pkgname}"` in [line 27 of the PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freecad-assembly3-git)?

ceremcem, 2021-01-23:
Yes, it seems `"${srcdir}/${pkgname}"` is `path/to/fc-build/Mod/asm3/freecad/asm3`. Just replace those lines in your code: 

```diff
build(){
-  cd "${srcdir}/${pkgname}"
+  asm3="${srcdir}/${pkgname}"
```

 (You'll force a full clone in place of a submodule, but I don't think it will cause any harm)

timmwille, 2021-01-23:
Ok I'm still having some bugs to fix and this is definitely not the final solution but with you help I think I found the cleanest solution I can easily remove and try differently :) thx, I'll share some updates when tested

**In hope it might help someone else as well*, this is the remix that I successfully used (did not get the PKGBUILD to work, might miss some coding skills for that and still not sure about this `"${srcdir}/${pkgname}"` part):

---
go to your home FreeCAD/Mod directory (where you FreeCAD Add-Ons are)

`cd ./FreeCAD/Mod` 

```bash
git clone https://github.com/realthunder/FreeCAD_assembly3.git asm3
cd asm3/freecad/asm3
git clone https://github.com/realthunder/solvespace.git slvs
cd slvs
git submodule update --init extlib/libdxfrw
mkdir -p build && cd build
cmake -DBUILD_PYTHON=1 -DPYTHON_EXECUTABLE:FILEPATH='/usr/bin/python3' ..
make _slvs
py_slvs_dir="../../py3_slvs"
mkdir -p $py_slvs_dir
cp src/swig/python/{slvs.py,_slvs.so} $py_slvs_dir
touch $py_slvs_dir/__init__.py
```
---

Side Note:
→ the only artefact I have to solve (but was having this before, might be to asm3 or my specific [AUR Build](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=freecad-linkstage3-git) and interface modification I'm testing): somehow some workbenches, like part, part design and also assembly 3, 4 and A2+ are now missing their Toolbars each time I restart FreeCAD (seems also to affect my Treeview adjustment, so I think it might not be asm3 directly, will check):

[image]

But yes I can easily activate with rightclick in the empty toolbar area and would call this for now a success, will focus on a bit more CAD design next, too much terminal code action the past hours :P 
[image]

luzpaz, 2022-07-01:
@ceremcem can you revisit this ticket ?