realthunder/FreeCAD_assembly3 #281: Linux AppImage crashes on deleting any object in tree view

anirudhnehra, 2019-11-21:
Hi,
When a failed or any other geometrical object is deleted in Tree view of the AppImage provided by you, FreeCAD 0.19 immediately crashes with SIGSEGV fault. It happens everytime when Delete key is pressed.

realthunder, 2019-11-21:
Are there any stack track print out after crash? Can you please paste it here. It would be better to provide a sample file causing the crash so that I can reproduce it. Thanks!

anirudhnehra, 2019-11-21:
Hi,
here is the terminal output

anirudh@anirudh:~/Desktop/freecad$ ./FreeCAD-asm3-Conda-Py3-Qt5-20190703-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA12UuXU
FreeCAD 0.19, Libs: 0.19R16830 +1496 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "inconsistent use of tabs and spaces in indentation (<string>, line 48)" occurred in /home/anirudh/.FreeCAD/Mod/GDT/InitGui.py
Please look into the log file for further information
<Import> ImportOCAF2.cpp(534): free shape count 10
<Import> AppImportGuiPy.cpp(519): file read time: 0.194313s
<Import> AppImportGuiPy.cpp(520): import time: 0.0391696s
<Import> AppImportGuiPy.cpp(521): total time: 0.233483s
54.012817 <asm3.sys> solver.py(352): no assembly found
PartDesign::LinearPattern / LinearPattern: Links go out of the allowed scope
PartDesign::LinearPattern / LinearPattern: Links go out of the allowed scope
PartDesign::LinearPattern / LinearPattern: Links go out of the allowed scope
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3a100) [0x7ffadfe9c100]
#1  0x7ffae23b8900 in Gui::DocumentItem::document() const from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0
#2  0x7ffae23d9592 in Gui::DocumentObjectItem::requiredAtRoot(bool) const from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x42
#3  0x7ffae23d86d4 in Gui::DocumentItem::populateItem(Gui::DocumentObjectItem*, bool, bool) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x484
#4  0x7ffae23d97c7 in Gui::TreeWidget::updateChildren(App::DocumentObject*, std::set<std::shared_ptr<Gui::DocumentObjectData>, std::less<std::shared_ptr<Gui::DocumentObjectData> >, std::allocator<std::shared_ptr<Gui::DocumentObjectData> > > const&, bool, bool) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0xe7
#5  0x7ffae23db9e1 in Gui::TreeWidget::onUpdateStatus() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x541
#6  0x7ffae23dc3c7 in Gui::TreeWidget::_updateStatus(bool) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x47
#7  0x7ffae23ddc49 in Gui::TreeWidget::onSelectTimer() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x19
#8  0x7ffae23deaa1 in Gui::TreeWidget::onItemSelectionChanged() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0xa1
#9  0x7ffae0394e34 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x780
#10  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Widgets.so.5(+0x333329) [0x7ffae091e329]
#11  0x7ffae0394e34 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x780
#12  0x7ffae0356e0c in QItemSelectionModel::selectionChanged(QItemSelection const&, QItemSelection const&) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x2a
#13  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5(+0x1867fe) [0x7ffae035a7fe]
#14  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5(+0x1886a0) [0x7ffae035c6a0]
#15  0x7ffae0394e34 in QMetaObject::activate(QObject*, int, int, void**) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x780
#16  0x7ffae034d460 in QAbstractItemModel::rowsAboutToBeRemoved(QModelIndex const&, int, int, QAbstractItemModel::QPrivateSignal) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x4e
#17  0x7ffae0350f57 in QAbstractItemModel::beginRemoveRows(QModelIndex const&, int, int) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Core.so.5+0x5d
#18  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Widgets.so.5(+0x3354ea) [0x7ffae09204ea]
#19  0x7ffae09205fe in QTreeWidgetItem::~QTreeWidgetItem() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libQt5Widgets.so.5+0x8a
#20  0x7ffae23dcfd7 in Gui::DocumentObjectItem::~DocumentObjectItem() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x117
#21  0x7ffae23e0df6 in Gui::TreeWidget::_slotDeleteObject(Gui::ViewProviderDocumentObject const&, Gui::DocumentItem*) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x1b6
#22  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so(+0x35a31c) [0x7ffae21b531c]
#23  0x7ffae220bb2b in Gui::Document::slotDeletedObject(App::DocumentObject const&) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x16b
#24  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADApp.so(+0x1a9e8c) [0x7ffae1b09e8c]
#25  0x7ffae1b204ae in App::Document::removeObject(char const*) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADApp.so+0x21e
#26  0x7ffae1b7b8cf in App::DocumentPy::removeObject(_object*) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADApp.so+0x6f
#27  0x7ffae1b7ba3a in App::DocumentPy::staticCallback_removeObject(_object*, _object*) from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADApp.so+0x3a
#28  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libpython3.7m.so.1.0(_PyMethodDef_RawFastCallKeywords+0x104) [0x7ffae1545034]
#29  /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libpython3.7m.so.1.0(_PyCFunction_FastCallKeywords+0x26) [0x7ffae1545ba6]
[... cut ...]
#61  0x7ffae21bbd04 in Gui::Application::runApplication() from /tmp/.mount_FreeCA12UuXU/usr/bin/../lib/libFreeCADGui.so+0x1534
#62  ./FreeCAD(main+0x725) [0x561de43797a5]
#63  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7ffadfe88bbb]
#64  ./FreeCAD(+0x44d5) [0x561de437a4d5]

Here is the freecad file used. It imports a STEP, then creates an assembly through Assembly3, then does some part design operations. If any feature say fillet(second last in tree) is deleted, the whole session crashes.
The freecad file used: https://drive.google.com/file/d/1iDUoLCgVSofWZruaGfyoP2mfhgxe9hku/view?usp=sharing

Thanks.

prof7bit, 2019-11-24:
I can reproduce this problem with the currently latest release: 

    FreeCAD-asm3-Conda-Py3-Qt5-20190703-glibc2.12-x86_64

Then I downloaded the second latest release:

    FreeCAD-asm3-Bionic-Py3n2-Qt5-20190619.glibc2.25-x86_64

And here it does not crash. I downloaded the "Bionic" build because I have Kubuntu Bionic and the latest release has no such build.

realthunder, 2019-11-24:
I can't reproduce it now. It is probably fixed since the release. I'll make a new release next month.

prof7bit, 2020-02-18:
> I'll make a new release next month.

Did you abandon the project? This would be very sad, we would all be doomed to spend another 20 years without topological naming in freecad (the single most important feature of your fork), and now this fork also seems to be dying. I am slowly beginning to lose all hope :-(

realthunder, 2020-02-18:
Wow wow wow, you are joking, right? I am working on it all the time. This `FreeCAD_assembly3` reop is the Python assembly workbench, not the `fork`. That will be [here](https://github.com/realthunder/FreeCAD), and the development is happening in [this](https://github.com/realthunder/FreeCAD/tree/LinkDev) branch.

I am working hard for the release this very moment. Sorry it got delayed for various reasons.

BTW, the topo naming feature will be available in upstream for the next release cycle, 0.20. Maybe they'll consider bumping the version to 1.0, not sure.

prof7bit, 2020-02-18:
Which repo or branch or pull request would I have to track where I could see the first signs of topo-naming merge being prepared (mostly to calm down my own impatience)? 

Sorry if I sounded too pessimistic, it was not my intention to criticize your hard work, I am just extremely impatient (this is my fault, not yours) and the single one most important thing I have been waiting for since like forever is topo naming and this repo is the only one that has an AppImage with topo naming. My greatest fear is that all this great work could be lost again if FreeCAD devs don't merge it because topo naming seems to be on a very low priority for them, everything else seems more important (which I really can't understand).

Sorry again if I sounded too impatient. Your work is very much appreciated.

realthunder, 2020-02-18:
Thanks for the appreciation. Right now the only repo have this feature is my fork. The current release candidate is 0.19, and it has been determined to not include the topo naming feature, due to the limited time remaining. I'll start submitting patches when 0.20 cycle starts, probably this April or May. You can visit FreeCAD github release page and check if 0.20 development build is online.

luzpaz, 2021-02-20:
@anirudhnehra can you reproduce on the latest asm3 appimage ? https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11

oficinerobotica, 2021-06-03:
Tested the example file with :
`OS: KDE neon User Edition 5.21 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.531.24301 +3702 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 4ad819f55a108b7e8bddb134d0de39218e662aa4
Python version: 3.8.10
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.4.0
Locale: English/United States (en_US)
`
and the issue can't be reproduced anymore.

closed 2021-06-03 by oficinerobotica