realthunder/FreeCAD_assembly3 #282: Spacenavigator does not work

MicroKloon, 2019-12-18:
My spacenavigator does no longer work in FreeCAD. It does not even show up in 
`Tools -> Customize -> Spaceball motion
`
Is this a problem with the ' oldish'  FreeCAD version bundled for asm3 (from june this year) or is there something else wrong? When I start up ' normal' FreeCAD the spacenavigator works as expected.

Using this version (asm3 version) it does not work:

```
OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16830 +1496 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 1b29767b545440b7ab93f1a5f2b742d622f1b547
Python version: 3.7.3
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Using the new version it does work:
```
OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18947 (Git) AppImage
Build type: Release
Branch: master
Hash: 91f6a5e5ee1b48063d5d4b59cc6250fc2a1021b1
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)
```

MicroKloon, 2019-12-19:
> Is it working before? I mean have you tried successfully with my asm3 June release? 

The june relase is the one that gives problems. I can not find an earlier (or later) version of your release with asm3. 
using: `FreeCAD-asm3-Conda-Py3-Qt5-20190703-glibc2.12-x86_64_cc1943c34fe980451062c6149df22cc8`
The 'normal'  FC release does work with the spacenavigator.

>Anyway, I plan to do a merge with the upstream and make another release this month. Hopefully it can be fixed by then.

Would be great to have a more current FC version!

Thank you for all your hard work, so nice to finally see an assembly solution that works and where you do not have to be a qualified engineer to understand what to do :)

anirudhnehra, 2020-02-05:
Hi,
It also does not work for me. However it sort of works with latest daily build of 0.19 but perfectly with 0.18 release.
Another issue is that if the mouse pointer is not on graphics screen( pointed in model tree, property view etc.) the 3d mouse does not work then. [This issue exists with daily build0.19 but not with stable 0.18] one has to click graphical window to reactivate 3d mouse inputs.
Again, it would be great to try the next release.
I am able to use external geometry in sketcher as"Defining Geometry" which is very useful. One more thing I would like to have is to be able to place Point On Curve constraint between a point and a bspline. Also, the sketcher automatically turns every external edge into a bspline projection, It would be great if sketcher recognizes lines as lines and arcs as arcs which it does sometimes but not every time especially with imported geometry.
Much Thanks

Anirudh

M4xTheMan, 2020-03-20:
I can confirm that it's not working with your latest release:
OS: Ubuntu 16.04.6 LTS (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16830 +1496 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 1b29767b545440b7ab93f1a5f2b742d622f1b547
Python version: 3.7.3
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

And it's working with this "normal" release:
OS: Ubuntu 16.04.6 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4.
Build type: Release
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

And it's working with this release too:
OS: Ubuntu 16.04.6 LTS (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.5.2
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

M4xTheMan, 2020-04-15:
I think I've found the answer to our problem. spacenavd is working in 0.19 but probably isn't part of the current Appimage for FreeCAD 0.19 including the Assembly3 workbench.

It'd be really great if we could get spacenavd support compiled into this Appimage! Check out this link for details: https://forum.freecadweb.org/viewtopic.php?f=8&t=37507&sid=8cca7d4d0f61349dd7a754b19e4e1bda&start=20

realthunder, 2020-04-24:
New image [released](../releases). Please download and try. BTW, any one of you use MacOS by any chance? I added libspnav to MacOS image as well, but obviously cannot test it myself.

M4xTheMan, 2020-04-24:
It works under Ubuntu 16.04.6 LTS 64-bit

Thank you very much!

luzpaz, 2021-02-03:
Can this issue now be closed? 
/cc @MicroKloon @anirudhnehra

snelweg, 2021-02-03:
It does work, but it still occasionally stops functioning after a crash. The spacenavd has to be restarted then.
But overall it does work, restarting FC several times a day is almost routine here.
Issue can be closed as far as I am concerned.

luzpaz, 2021-02-03:
@snelweg Sorry to hear that. Do you get the same behavior on the official FreeCAD master as well? Please report the other crashes then so we can address them.

snelweg, 2021-02-03:
> @snelweg Do you get the same behavior on the official FreeCAD master as well? 

I am not using the official branch, switching between Linkstage and official gave me so much trouble. And I like Linkstage better, it is more robust and less crashes.

>Please report the other crashes then so we can address them.

I will

joanperelopez, 2021-02-12:
Spacenav is not detected in MacOSX Freecad real thunder. The original 0.19 it does.

joanperelopez, 2021-02-12:
Someone knows some solution?

DIYman, 2021-02-24:
MacOSx customize menu says no spaceball present for me as well and it doesnt work.

Perhaps in the future as more features come in we can use it, I watched RTs interview with OficineRobotica on youtube really excited about the future of this branch.

luzpaz, 2021-02-24:
@realthunder did you pull in the changes to the 3DConnexion/spacemouse  code in master recently? 
https://github.com/FreeCAD/FreeCAD/search?q=3dconnexion+OR+spacemouse&type=commits

joanperelopez, 2021-02-24:
@realthunder I am working on MAC. I the actual version 0.11 it doesn't appear the text "space mouse not detected" instead show all the space mouse configuration window bars and options, but even that it seams to be correct the space mouse is not working at all.

realthunder, 2021-02-24:
> @realthunder did you pull in the changes to the 3DConnexion/spacemouse code in master recently?

The latest release is merged with upstream 02.07. I don't think there is any critical SpaceMouse fix. This is most likely a build issue on my side. My purchase of SpaceMouse is on the way. I'll fix the problem soon.

joanperelopez, 2021-02-25:
@realthunder Many thanks. I am waiting to use this branch when fixed instead of the official version.

realthunder, 2021-03-07:
Related to #471

joanperelopez, 2021-03-08:
@realthunder Great news!!  I work on MAC and this new release the SpaceMouse is working. Is not so fluid as Freecad 0.19, but is possible work. Many thanks.

DIYman, 2021-03-08:
@realthunder Thanks a lot, lightning reaction speed for a would be nice to have feature :) 

@joanperelopez Problems with its fluidity might be due to the recent MAC drivers coming out of 3dconnexion, they didnt release a stable version for Big Sur yet, it is still at beta, they are not as fast as Real Thunder :) It might come out soon though.

luzpaz, 2022-03-21:
Can we close this ticket?

closed 2022-03-22 by realthunder