realthunder/FreeCAD_assembly3 #285: "No module named assembly" error for LinkDev

ceremcem, 2020-02-16:
I built LinkDev branch and when I try to open my existing project, I'm getting the following errors many time during opening (like 1000 times?): 

```
<string>(2)<type 'exceptions.ImportError'>: No module named assembly
```

When the project is opened, latest errors and warnings are as follows: 

```
<string>(2)<type 'exceptions.ImportError'>: No module named assembly
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem013
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem014
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem016
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem013
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem014
SE::getSourceShapes - source shape is empty!
DVP::execute - source shape is invalid - (but document is restoring) - ProjItem016
```

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21152 (Git)
Build type: Release
Branch: LinkDev
Hash: 29c3b75829a1e49c05d6b304200b61451306bddf
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-02-16:
Can you check if you have asm3 installed in more than one place? Forgot to mention, new asm3 directory structure requires you to install the module into your `.FreeCAD/Mod` directory, instead of `Ext`.

There are quite drastic changes in upstream TechDraw lately. I'll take a look.

ceremcem, 2020-02-16:
I moved `asm3` to `~/.FreeCAD/Mod` and it worked:

```
mv /opt/FreeCAD/Ext/freecad/asm3/ /home/aea/.FreeCAD/Mod/
```

ceremcem, 2020-02-16:
There were couple of warnings and after a Refresh, there are some errors: 

```
<PropShape> PropertyTopoShape.cpp(359): Recomputation required for document 'driver_base' on geo element version change: 1.14.70200.3.0 -> sm1.1.14.70200.3.0
Warning - no numeric format in formatSpec  - Dimension006
Warning - no numeric format in formatSpec  - Dimension006
Warning - no numeric format in formatSpec  - Dimension007
Warning - no numeric format in formatSpec  - Dimension007
Warning - no numeric format in formatSpec  - Dimension008
Warning - no numeric format in formatSpec  - Dimension008
Warning - no numeric format in formatSpec  - Dimension008
Warning - no numeric format in formatSpec  - Dimension008
Warning - no numeric format in formatSpec  - Dimension007
Warning - no numeric format in formatSpec  - Dimension007
Warning - no numeric format in formatSpec  - Dimension006
Warning - no numeric format in formatSpec  - Dimension006
Warning - no numeric format in formatSpec  - Dimension
Warning - no numeric format in formatSpec  - Dimension
Warning - no numeric format in formatSpec  - Dimension002
Warning - no numeric format in formatSpec  - Dimension002
Warning - no numeric format in formatSpec  - Dimension003
Warning - no numeric format in formatSpec  - Dimension003
Warning - no numeric format in formatSpec  - Dimension004
Warning - no numeric format in formatSpec  - Dimension004
Warning - no numeric format in formatSpec  - Dimension005
Warning - no numeric format in formatSpec  - Dimension005
Warning - no numeric format in formatSpec  - Dimension009
Warning - no numeric format in formatSpec  - Dimension009
Warning - no numeric format in formatSpec  - Dimension010
Warning - no numeric format in formatSpec  - Dimension010
Warning - no numeric format in formatSpec  - Dimension010
Warning - no numeric format in formatSpec  - Dimension010
Warning - no numeric format in formatSpec  - Dimension009
Warning - no numeric format in formatSpec  - Dimension009
Warning - no numeric format in formatSpec  - Dimension005
Warning - no numeric format in formatSpec  - Dimension005
Warning - no numeric format in formatSpec  - Dimension004
Warning - no numeric format in formatSpec  - Dimension004
Warning - no numeric format in formatSpec  - Dimension003
Warning - no numeric format in formatSpec  - Dimension003
Warning - no numeric format in formatSpec  - Dimension002
Warning - no numeric format in formatSpec  - Dimension002
Warning - no numeric format in formatSpec  - Dimension
Warning - no numeric format in formatSpec  - Dimension
<Exception> <string>(1)|(0): Failed to convert to Quantity
in property binding 'driver_base#Constraint093.Label2'
131.622 <App> <string>(1)|Document.cpp(3712): Failed to recompute driver_base#Constraint093: Failed to convert to Quantity
in property binding 'driver_base#Constraint093.Label2'
<Exception> <string>(1)|(0): Failed to convert to Quantity
in property binding 'driver_base#Constraint178.Label2'
132.262 <App> <string>(1)|Document.cpp(3712): Failed to recompute driver_base#Constraint178: Failed to convert to Quantity
in property binding 'driver_base#Constraint178.Label2'
Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 613, in execute
    self.getElementSubname())
  File "/home/aea/.FreeCAD/Mod/asm3/freecad/asm3/assembly.py", line 1374, in getElementInfo
    '{}.{}'.format(objName(part),subname))
<type 'exceptions.RuntimeError'>: Failed to get geometry element from driver_base#Body010.Bend003.;#1417;:T5825:5:E.Edge52
Recompute failed! Please check report view.
```

ceremcem, 2020-02-16:
When I wanted to remove the complaining constraint, FreeCAD crashed: 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7f15f8292060]
#1  0x7f15ff1120da in SoBase::isOfType(SoType) const from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x1a
#2  0x7f1601c5dd59 in Gui::SoFCPathAnnotation::setPath(SoPath*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x59
#3  0x7f1601c5df69 in Gui::SoFCPathAnnotation::~SoFCPathAnnotation() from /opt/FreeCAD/lib/libFreeCADGui.so+0x19
#4  0x7f1601c5dfa9 in Gui::SoFCPathAnnotation::~SoFCPathAnnotation() from /opt/FreeCAD/lib/libFreeCADGui.so+0x9
#5  0x7f15ff10f172 in SoBase::destroy() from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0x102
#6  0x7f15ff111e12 in SoBase::unref() const from /usr/lib/x86_64-linux-gnu/libCoin.so.80+0xa2
#7  0x7f1601d21b34 in Gui::View3DInventorViewer::OnTopInfo::~OnTopInfo() from /opt/FreeCAD/lib/libFreeCADGui.so+0x14
#8  0x7f1601d2fa0c in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, Gui::View3DInventorViewer::OnTopInfo>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, Gui::View3DInventorViewer::OnTopInfo> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::erase(std::__detail::_Node_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, Gui::View3DInventorViewer::OnTopInfo>, false, true>) from /opt/FreeCAD/lib/libFreeCADGui.so+0x6c
#9  0x7f1601d27695 in Gui::View3DInventorViewer::checkGroupOnTop(Gui::SelectionChanges const&, bool) from /opt/FreeCAD/lib/libFreeCADGui.so+0xb05
#10  0x7f1601d28ab1 in Gui::View3DInventorViewer::onSelectionChanged(Gui::SelectionChanges const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1c1
#11  0x7f1601e6f0e1 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x21
#12  0x7f1601bee951 in boost::signals2::detail::signal_impl<void (Gui::SelectionChanges const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)>, boost::function<void (boost::signals2::connection const&, Gui::SelectionChanges const&)>, boost::signals2::mutex>::operator()(Gui::SelectionChanges const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1c1
#13  0x7f1601e77ef5 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x125
#14  0x7f1601e78856 in Gui::SelectionSingleton::rmvPreselect(bool) from /opt/FreeCAD/lib/libFreeCADGui.so+0x4c6
#15  0x7f1601e7a633 in Gui::SelectionSingleton::slotDeletedObject(App::DocumentObject const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x43
#16  0x7f160103bef1 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&) from /opt/FreeCAD/lib/libFreeCADApp.so+0x1c1
#17  0x7f160122edf0 in App::Application::slotDeletedObject(App::DocumentObject const&) from /opt/FreeCAD/lib/libFreeCADApp.so+0x10
#18  0x7f160103bef1 in boost::signals2::detail::signal_impl<void (App::DocumentObject const&), boost::signals2::optional_last_value<void>, int, std::less<int>, boost::function<void (App::DocumentObject const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&) from /opt/FreeCAD/lib/libFreeCADApp.so+0x1c1
#19  0x7f1600ffa6e9 in App::Document::removeObject(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x1f9
#20  0x7f160114db5e in App::LinkBaseExtension::detachElement(App::DocumentObject*) from /opt/FreeCAD/lib/libFreeCADApp.so+0xce
#21  0x7f160114ee5b in App::LinkBaseExtension::onExtendedUnsetupObject() from /opt/FreeCAD/lib/libFreeCADApp.so+0x8b
#22  0x7f1601051b93 in void App::ExtensionContainer::callExtension<void (App::DocumentObjectExtension::*)()>(void (App::DocumentObjectExtension::*&&)()) const from /opt/FreeCAD/lib/libFreeCADApp.so+0x83
#23  0x7f160104dd4d in App::DocumentObject::unsetupObject() from /opt/FreeCAD/lib/libFreeCADApp.so+0x1d
#24  0x7f1600ffa8b4 in App::Document::removeObject(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x3c4
#25  0x7f16010965ef in App::DocumentPy::removeObject(_object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x5f
#26  0x7f1601096750 in App::DocumentPy::staticCallback_removeObject(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADApp.so+0x40
#27  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8541) [0x7f15ffd36091]
#28  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f15ffe9d29c]
#29  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f15ffd2d9c9]
[... cut ...]
#58  0x7f15f9683b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#59  0x7f15f9689bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#60  0x7f16019a8e8e in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x1a6e
#61  freecad-git(main+0x8aa) [0x560ec1eb284a]
#62  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f15f827f2e1]
#63  freecad-git(_start+0x2a) [0x560ec1eb399a]
```

closed 2020-02-16 by ceremcem

realthunder, 2020-02-16:
Can you please send me the file that cause the crash, and tell me which object you deleted? Thanks

ceremcem, 2020-02-16:
It's probably the same project will cause the same crash that I previously sent, but I'm handing it over via FC DM, just in case.

ceremcem, 2020-02-16:
Reproduction: 

1. Open `button_driver.fcstd`
2. Refresh
3. See the errors in Report View 
4. Delete `driver_base/housing/Constraints/bend_outer_distance` by clicking on it and pressing `DEL` button. 
5. Interestingly this caused the crash 2 times, however I can not reproduce the crash again.

realthunder, 2020-02-16:
Did some quick test, no crash. I'll take a close look tomorrow.

realthunder, 2020-02-17:
I have reproduced the crash. It should be fixed now, along with some other fixes. Please sync both LinkDev and asm3 before trying.