realthunder/FreeCAD_assembly3 #288: Exception: Array.LinkCopyOnChange already exists

ceremcem, 2020-05-13:
After updating FC and Asm3, I opened my previous project. A popup opened and declared a required recomputation. After recomputation, everything seemed to be fine. 

After closing and opening the project, an exception is thrown in the Report View: 

```
Traceback (most recent call last):
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2619, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2746, in linkSetup
    _DraftLink.linkSetup(self,obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2588, in linkSetup
    obj.addProperty("App::PropertyEnumeration","LinkCopyOnChange"," Link")
<type 'exceptions.RuntimeError'>: Property MC_1_5_6_G_3_81#Array.LinkCopyOnChange already exists
Traceback (most recent call last):
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2619, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2746, in linkSetup
    _DraftLink.linkSetup(self,obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2588, in linkSetup
    obj.addProperty("App::PropertyEnumeration","LinkCopyOnChange"," Link")
<type 'exceptions.RuntimeError'>: Property MC_1_5_4_G_3_81#Array.LinkCopyOnChange already exists
Traceback (most recent call last):
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2619, in onDocumentRestored
    self.linkSetup(obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2746, in linkSetup
    _DraftLink.linkSetup(self,obj)
  File "/opt/FreeCAD/Mod/Draft/Draft.py", line 2588, in linkSetup
    obj.addProperty("App::PropertyEnumeration","LinkCopyOnChange"," Link")
<type 'exceptions.RuntimeError'>: Property tb6600_module#Array.LinkCopyOnChange already exists
QGISVGTemplate::createClickHandles - xml loading error
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23146 (Git)
Build type: Release
Branch: LinkDev
Hash: 52fba79b3289e6d3c010b6aac4213f35ae418c1e
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

(Link sent via Forum DM)

ceremcem, 2020-05-13:
Update: 

The very similar issue is thrown as a warning in my previous version of FC: 

```
<PropShape> PropertyTopoShape.cpp(359): Recomputation required for document 'driver_base' on geo element version change: 1.14.70200.3.0 -> sm1.1.14.70200.3.0
QGISVGTemplate::createClickHandles - xml loading error
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21555 (Git)
Build type: Release
Branch: LinkDev
Hash: bbde42384d60e98a3f76f1b21f1b7a0e7aa8f382
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

realthunder, 2020-05-14:
Fixed. Please try.

closed 2020-05-14 by realthunder

ceremcem, 2020-05-14:
No problem in this version (ebd99b956e25d340195f4ca12355f8ada7bd3853), thanks.