realthunder/FreeCAD_assembly3 #289: FreeCAD Crash

ceremcem, 2020-05-13:
When I add a dimension measurement in TechDraw page and then delete it, FreeCAD crashes with the following traceback: 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7f3cc9e290c0]
#1  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fb44a) [0x7f3cc6ca644a]
#2  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fa065) [0x7f3cc6ca5065]
#3  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fa065) [0x7f3cc6ca5065]
#4  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fa065) [0x7f3cc6ca5065]
#5  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fa011) [0x7f3cc6ca5011]
#6  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fa0ff) [0x7f3cc6ca50ff]
#7  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fd66b) [0x7f3cc6ca866b]
#8  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7fd877) [0x7f3cc6ca8877]
#9  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7ef61e) [0x7f3cc6c9a61e]
#10  0x7f3cc6cb9e8a in QGraphicsView::paintEvent(QPaintEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x4fa
#11  0x7f3c36623f6a in TechDrawGui::QGVPage::paintEvent(QPaintEvent*) from /opt/FreeCAD/lib/TechDrawGui.so+0x1a
#12  0x7f3cc66c5e80 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x290
#13  0x7f3cc6a8ac8e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e
#14  0x7f3cc6cb8b74 in QGraphicsView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x184
#15  0x7f3cc5ba90a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#16  0x7f3cc666f52c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#17  0x7f3cc667652c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#18  0x7f3ccdf79a41 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#19  0x7f3cc5ba8f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#20  0x7f3cc66c04e5 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x605
#21  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x3e55f0) [0x7f3cc68905f0]
#22  0x7f3cc66b31e0 in QWidgetPrivate::syncBackingStore() from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xa0
#23  0x7f3cc66c5f68 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x378
#24  0x7f3cc6aa5803 in QMainWindow::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x113
#25  0x7f3cce33e854 in Gui::MainWindow::event(QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x64
#26  0x7f3cc666f54c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#27  0x7f3cc667652c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#28  0x7f3ccdf79a41 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#29  0x7f3cc5ba8f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#30  0x7f3cc5baca16 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3b6
#31  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7e24d2) [0x7f3cc6c8d4d2]
#32  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x7e6119) [0x7f3cc6c91119]
#33  0x7f3cc5bc3381 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1d1
#34  0x7f3cc6ca02ab in QGraphicsScene::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x49b
#35  0x7f3cc666f54c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#36  0x7f3cc667652c in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2cc
#37  0x7f3ccdf79a41 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x41
#38  0x7f3cc5ba8f1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#39  0x7f3cc5baca16 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3b6
#40  /usr/lib/x86_64-linux-gnu/libQtCore.so.4(+0x1bb703) [0x7f3cc5bd9703]
#41  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f3cbd13e7f7]
#42  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f3cbd13ea60]
#43  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f3cbd13eb0c]
#44  0x7f3cc5bd9854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#45  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f3cc67195d6]
#46  0x7f3cc5ba77ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#47  0x7f3cc5ba7b55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#48  0x7f3cc5badbd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#49  0x7f3ccdee965e in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x1ace
#50  freecad-git(main+0x8aa) [0x55c1333b484a]
#51  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f3cc47a32e1]
#52  freecad-git(_start+0x2a) [0x55c1333b599a]
```

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21555 (Git)
Build type: Release
Branch: LinkDev
Hash: bbde42384d60e98a3f76f1b21f1b7a0e7aa8f382
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachment 

(Sent via Forum DM)

realthunder, 2020-05-14:
I can't reproduce it. Can you be more specific? Like which document and object, which edge, and which dimension you have tried.

ceremcem, 2020-05-14:
Interesting. I opened my project, navigated to `driver_base/fabrication_output/2mm_metal`, selected the top left drill hole of bottom right square shape, added a "diameter measurement", it said "it's an ellipse", I said "ok", added a diameter measurement entity, then delete it, then FreeCAD crashed. I wanted to provide a screencast but this time it doesn't crash.

ceremcem, 2020-05-14:
Okay, I made it happen: 

[image]

# Attachment 

I'm sending the latest version of my project just in case. (Forum DM)

ceremcem, 2020-05-14:
FYI: I can not reproduce the crash in version `ebd99b956e25d340195f4ca12355f8ada7bd3853` and it "looks like" it won't crash. 

(In the previous version of FC, deletion took a couple of seconds. However, in this version, it deletes the entity instantly, so I guess it won't crash in this version)

ceremcem, 2021-02-14:
Closing regarding to my previous post.

closed 2021-02-14 by ceremcem