realthunder/FreeCAD_assembly3 #290: Crash while saving dependent files

yrabbit, 2020-05-24:
I use AppImage 2020.05.22
I apologize for the big model, but I can’t understand what I'm doing wrong and I can’t repeat on simple models.

[image]

call trace:
[report.txt](https://github.com/realthunder/FreeCAD_assembly3/files/4672761/report.txt)
project:
[example.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/4672763/example.tar.gz)

lilaL, 2020-05-25:
I can save your file with 05.22 but I get the following message:
```
82.8627 <App> Document.cpp(3057): Swapboard#Link.LinkedObject: Link not restored
Linked object: Assembly
Linked file: PCBSwapboard.FCStd
82.8793 <App> Document.cpp(3057): Swapboard#Link002.LinkedObject: Link not restored
Linked object: Assembly
Linked document: Plug64
```
So I guess something is wrong with your dependencies

realthunder, 2020-05-25:
I have already fixed the problem, and also resolve the 'Link not restored' the error. I'll make a release tomorrow. Thanks for reporting.

closed 2020-05-25 by realthunder

yrabbit, 2020-05-25:
> I can save your file with 05.22 but I get the following message:
>
> ```
> 82.8627 <App> Document.cpp(3057): Swapboard#Link.LinkedObject: Link not restored
> Linked object: Assembly
> Linked file: [...]

I did not see these messages because in the settings I have set to always download all documents completely.
I wonder if the situation will repeat for you with this setting?

> 
> So I guess something is wrong with your dependencies

As an option, I will have to repeat the creation of all models from scratch constantly checking. Although I tried very hard not to make cyclic dependencies:(

[image]

realthunder, 2020-05-26:
Yes, the message does depend on this 'Partial loading' setting. It is a bit difficult to explain. Freezing the assembly allows the document to be partially loaded when linked, which means none of its parts needs to be loaded. However, there is a runtime check to see if there is any touched objects after partial loading a document. If found, the document must be reloaded in full. There is a bug in handling of this reload process, causing the missing link message. I have already fixed this bug.

There is no cyclic dependency in you file as far as I can see.

yrabbit, 2020-05-26:
Thank you very much!
I hope that with the new AppImage, I can "Recompute -> Save" (with all dependent files). The ability to save a file is so important to me that I can unfreeze everything if necessary and live with it:)

realthunder, 2020-05-26:
Release image updated. Please try.

yrabbit, 2020-05-27:
Aye! 
It was possible to open/recalculate without problems and, most importantly, save not only this example (and it is not my largest), but also other projects. Thanks!

I will also try to return to storing files as directories (I had to refuse because there the consequences of failures during saving were much more serious).

realthunder, 2020-05-27:
> I will also try to return to storing files as directories (I had to refuse because there the consequences of failures during saving were much more serious).

Saving to directory are meant to be used with an external version control software. Take `git` for example. Very time you made a save, you can use `git status` to find out the new/removed/modified files, and do a commit. I will implement some automation of these steps in the future.

yrabbit, 2020-06-02:
Yes, surely automating the creation of commits with each save would help. You see, it is now possible to save the project, but it may not be possible to read back: after I received the "The graph must be a DAG" message a couple of times and had to go back to hellishly old archive, I do commit after each save manually.
Perhaps if (optionally) a dependency graph was built and checked with each record, I would feel more confident.