realthunder/FreeCAD_assembly3 #291: MacOS FreeCAD issue in freecad.asm3 at startup

zaphodbe, 2020-05-27:
At startup from double clicking on icon or from command line I get the following error on startup.  Seems like it is trying to shut down FreeCAD.  Then lots of gl_context errors.  Seems to be able to open and edit documents.

Using the latest 0.11 2020.05.26 and the previous 5.22 versions.  
An empty start screen is shown.

`./FreeCAD
FreeCAD 0.19, Libs: 0.19R20886 +2302 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
During initialization the error "invalid syntax (process.py, line 262)" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 150, in InitApplications
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/asm3/freecad/asm3/init_gui.py", line 9, in <module>
    from . import sys_sympy
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/asm3/freecad/asm3/sys_sympy.py", line 12, in <module>
    import scipy.optimize as sopt
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.7/site-packages/scipy/optimize/__init__.py", line 389, in <module>
    from .optimize import *
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.7/site-packages/scipy/optimize/optimize.py", line 40, in <module>
    from scipy._lib._util import getargspec_no_self as _getargspec
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.7/site-packages/scipy/_lib/_util.py", line 9, in <module>
    from multiprocessing import Pool
  File "/Users/ccoley/Library/Preferences/FreeCAD/Mod/retr3d/multiprocessing/__init__.py", line 64, in <module>
    from multiprocessing.process import Process, current_process, active_children
  File "/Users/ccoley/Library/Preferences/FreeCAD/Mod/retr3d/multiprocessing/process.py", line 262
    except SystemExit, e:
                     ^
SyntaxError: invalid syntax
--------------------------------------------------------------------------------
[72468:58371:0526/222115.289806:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.289859:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.290604:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.290623:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.291111:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.291127:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.291547:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.291564:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.291958:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.291974:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.292370:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.292387:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.292782:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.292798:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.293188:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.293205:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.293587:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.293602:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
[72468:58371:0526/222115.293985:ERROR:gl_context_cgl.cc(136)] Error creating context.
[72468:58371:0526/222115.294000:ERROR:gles2_command_buffer_stub.cc(262)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
`

realthunder, 2020-05-27:
There are two errors according to your captured log. First is Python syntax error. Although it is reported by asm3, it is actually caused by the retr3d module you installed in the user directory. This module bundles an out-dated Python system module `multiprocessing`. Please try update retr3d, or remove it and try again.

The other OpenGL error seems to be related to [this](https://bugreports.qt.io/browse/QTBUG-73799) qt bug. It should only affect the `Start` module. You'll probably be just fine without using it.

zaphodbe, 2020-05-27:
Ok that explains my confusion.  Was thinking it was asm3 based on the first message.  Removed retr3d and that has gone.  Thanks.

Trying to figure out how to correct the gl_context error.  Another machine is not exhibiting this issue and the start screen displays correctly.  I assume it is something to do with which version of qt is installed. 

Is there a way to test which Qt is being used?
The referenced bug says to pass  --ignore-gpu-blacklist to qtwebengine applications to fix it but freecad won't accept that option.

luzpaz, 2021-02-20:
@zaphodbe if you look in the `Help > About FreeCAD` it should tell you what Qt version it's compiled against.

luzpaz, 2021-06-26:
No response from OP. This is an upstream Qt issue per https://github.com/realthunder/FreeCAD_assembly3/issues/291#issuecomment-634451434  
Closing ticket

closed 2021-06-26 by luzpaz