realthunder/FreeCAD_assembly3 #292: Point in Part/PartDesign gets complaints

parnasist3, 2020-06-05:
When creating a point in Part or PartDesign I get complaints, see attached files.
This is also true when creating only points in a sketch and "publishing" them.

hmm, seems github does not accept *.txt-files or *.png-files...
Any other way of attaching files here?

realthunder, 2020-06-05:
Just append the file with `.zip`. A screenshot may also help.

parnasist3, 2020-06-06:
OS: openSUSE Leap 15.1 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21300 +2313 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 0f809d2f8f3b8603c0310ffcfb13869710a0ec48
Python version: 3.7.6
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0

Screen dumps in each *.zip-file.

[Part.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4740604/Part.zip)
[PartDesign.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4740605/PartDesign.zip)

realthunder, 2020-06-08:
You are trying to attach a point to an `Origin` object. Despite its name, `Origin` is not a point. And as the error message suggests, `Origin` is not a geometry feature either. Depending on your use case, you can simply attach the point to the parent group (either body or part) of the `Origin`.

Note that you do not want to attach a point to its owner group, because that doesn't make any sense. The point moves when its owner geo group (body, part, etc) moves. You don't need to attach to it to get this behavior.

parnasist3, 2020-06-23:
Okey, so for now there is no way to create a point with respect to a fixed coordinate system or a fixed point?

But just because a fixed coordinate system and or fixed point doesn't exist as of now, it will not stop you from creating it?

I hope you recognize the very core functionality of each file having a fixed, non removable, coordinate system and a fixed point as an absolute/unquestionable reference to everything else in the part? We can call it absolute zero for the sake of simplicity.

If you do not fully understand how CAD in the real world/companies works, just ask and I'll teach you.

parnasist3, 2020-09-12:
OS: openSUSE Leap 15.1 (KDE//usr/share/xsessions/default)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21911 +2545 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 04c80c99eee0f7d2f761bdc7d8b32c077adefdec
Python version: 3.7.8
Qt version: 5.12.1
Coin version: 4.0.0
OCC version: 7.4.0

I'm missing:

 **Project point on line/curve/plane/surface**
    _Creation mode:_ normal to line/plane/surface.
     **or**
    _Creation mode:_ shortest distance to curve.
     **or**
    _Creation mode:_ along a direction to a plane/surface.

and

 **Create point/points on line/curve**
    _Creation mode:_ select line/curve, then an already existing point on that line/curve then direction along the selected line/curve, then real/relative distance between them. If real distance would overshoot the length of the line/curve, then use the tangential/curvature continuation of the line/curve to fulfill the command.

and

 **Create point on plane/surface**
    _Creation mode:_ select plane/surface, then an already existing point on that plane/surface then U+V which are projections of the current in work coordinate system or a another coordinate system of choice [one can have many different coordinate system in a part but only one is effective at the time]. If real distance would overshoot the length of the surface, then use the tangential/curvature continuation of the surface to fulfill the command.

and

 **Create point/points between two other points**
    _Creation mode:_ select two already existing points and set one as master/start. Middle point = relative value 0,5 relative to the two original points. Multiple points = equal spacing relative to the two original points.

and

**Create plane through a line/axis and angled with respect to a plane**
    _Creation mode:_ select the rotation line/axis the plane shall pass through, then an already existing plane or a plane of a current in work coordinate system or a another coordinate system of choice as a reference. In the background: always project the line/axis on the reference plane to get the correct reference angle, then translate the newly created plane back to the line/axis. As already stated this projection and translation shall not be seen by the user.

and

**Flip normal of plane/surface + flip direction of line/curve**
   _Creation mode:_ select plane/surface or line/curve, now an arrow in the 3D-viewport shows its current normal orientation. Clicking on the "flip-button" performs the "flip" and the arrow in the 3D-viewport shows its new orientation. The "flip" shall be visible in the tree as separate icon just below what was "flipped". Hovering over the icon shall highlight the original  plane/surface or line/curve in the tree and in the 3D-viewport.

and

**Replace command for points, lines, curves, planes, surfaces and bodies**
   _Creation mode:_ set "Tip" on/for the feature/entity to be replaced. Right-click to select "Replace with..." Click on the new feature/entity. In the background (especially for bodies) check if the numbers of vertexes + edges + surface are the same and in the, roughly, same position. If any of the checks fails a popup window will show two sand-boxed 3D-viewports, one with the current feature/entity to be replaced and one with the new feature/entity.
In the left view-port one shall be presented with the vertexes, edges and surfaces that the replace-check couldn't find an good enough replacement for. They shall be numbered and colored in such a way they are easily seen and of such a size they are easy to click on.
The clicked feature/entity will then be replaced by whatever the user clicks on in the right 3D-viewport.
Both view-ports shall be linked in such a way that zooming, panning and rotating is 1:1.
Nothing will be sent to the tree before the user is happy with the result, which is shown in the right view-port, a sort of preview so to say. When the user is happy he/she clicks on the accept-button.
The whole model shall not, by default, be updated, to avoid unnecessary crashes.

and

**Re-linking of features/entity's**
   _Creation mode:_ an update of the model comes to a halt somewhere in the model, then the "Re-linking" window pops up doing the same thing as the above mentioned "sand-boxed dual 3D-viewport".

For all the above the point shall of course stay connected in such a way that when one change position of the line/curve/plane/surface the point/points follows.

Your comments?

realthunder, 2020-09-14:
That's quite a feature request. Yes, I do plan to work on projection tools, but I am busy with something else at the moment. I suggest you post this feature request to the forum to gather more opinions. And someone else may be interested and offer to implement it right away.

parnasist3, 2020-09-14:
I'm only glad you didn't chuck it out of the window.
Now I at least know you're not opposing these feature requests.
For now I will cross my fingers and hope it will find its way into, at least, your fork sometime in the future.

parnasist3, 2020-11-13:
I've just tested your 2020.11.12.22739 +2746 (Git) AppImage and noted you've added the possibility to create points and coordinate systems outside Body without complaints when assigning the point to "Objects origin" !
Very nice and thank you for that!

luzpaz, 2022-07-04:
How do we re-word this ticket as a Feature request then?