realthunder/FreeCAD_assembly3 #293: AppImage since  do no run on Fedora 31

paullee0, 2020-06-06:
```
Downloads]$ ./FreeCAD-asm3-Conda-Py3-Qt5-20190703-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA2JDiHA
./FreeCAD: error while loading shared libraries: libnsl.so.1: cannot open shared object file: No such file or directory
```

FreeCAD-asm3-Conda-Py3-Qt5-20190703-glibc2.12-x86_64.AppImage
FreeCAD-asm3-Conda-Py3-Qt5-20200522-glibc2.12-x86_64.AppImage
FreeCAD-asm3-Conda-Py3-Qt5-20200604-glibc2.12-x86_64.AppImage
- same result :)

realthunder, 2020-06-06:
Try the solution [here](https://github.com/AppImage/pkg2appimage/issues/336#issuecomment-419797608)?

paullee0, 2020-06-06:
Thanks for the quick response !

libnsl2.x86_64 is found already installed
also install libnsl2-devel.x86_64 ... now.

Same error message :( ...

realthunder, 2020-06-06:
I believe the missing library is the legacy libnsl1, not the newer libnsl2?

paullee0, 2020-06-06:
Oh I see, i am not reading that post carefully :)

It seem running... many thanks !

ceremcem, 2021-02-21:
Closing due to author's comment.

closed 2021-02-21 by ceremcem