realthunder/FreeCAD_assembly3 #294: AppImage Crashing on opening 0.18 freecad files

anirudhnehra, 2020-06-08:
Hello,
I am trying to open designs created with 0.18 version of freecad. The software crashes on opening the files.
I attach a sample document. Interestingly, the file opens without issue in stable 0.19 release of freecad. Please suggest what to do.
[mono_final_2.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4746036/mono_final_2.FCStd.zip)

realthunder, 2020-06-08:
Can you please attach the FreeCAD about information here? I just tried the Linux AppImage and Windows image. Both works fine.

anirudhnehra, 2020-06-08:
Hi, where do I get freecad Info?
I use your latest AppImage release build 
FreeCAD-asm3-Conda-Py3-Qt5-20200509-glibc2.12-x86_64.AppImage

realthunder, 2020-06-08:
Please try with the latest [release](https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.11). I update the image quite frequently.

The `About` information can be obtained at main menu -> Help -> About FreeCAD.

anirudhnehra, 2020-06-09:
Thank you its working now.

closed 2020-06-09 by anirudhnehra