realthunder/FreeCAD_assembly3 #295: Draft entities do not snap to vertices/floating in space

anirudhnehra, 2020-06-09:
Hi,
When I use the latest Appimage, and try using draft entities such as line, bspline etc, the lines do not snap to selected vertices and are kind of floating in space in random orientation. I attach a screen capture. I tried turinng off/on all snap options.

Thanks
Anirudh
[snapping_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4749285/snapping_issue.zip)

anirudhnehra, 2020-06-09:
Again an issue with opening old models craeted with Aseembly 3 Appimage. Please see this file. I a unable to open it in the latest AppImage of Assembly3.It does open sometimes but with it almost freezes the system.
[big_sketch.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4750207/big_sketch.zip)

realthunder, 2020-06-09:
> Again an issue with opening old models craeted with Aseembly 3 Appimage. Please see this file. I a unable to open it in the latest AppImage of Assembly3.It does open sometimes but with it almost freezes the system.

There is probably something wrong with OCCT 7.4. You can try upstream AppImage. It is the same. OCCT 7.4 must have done something to the triangulation algorithm. The freeze is caused by excessive optimization due to extreme `AngularDeflection` setting. Change it back to the default 28.5 will allow you to load the file.

I'll look into the draft snapping issue later.

anirudhnehra, 2020-06-09:
Yes, it does work However, the upstream development version AppImage (standard Freecad) is able to load the model without any issue.

anirudhnehra, 2020-06-09:
OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18846 (Git) AppImage
Build type: Release
Branch: master
Hash: ceeb776fff25c679752571d5d8e6c4c8b1fd9008
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/India (en_IN)

realthunder, 2020-06-09:
> OCC version: 7.3.0

Because you are using an older build of the upstream. The latest upstream AppImage is also using OCC 7.4, which has the same problem.

If you have a Windows machine, you can try my Windows release, it is still using OCC 7.3.

anirudhnehra, 2020-06-11:
Its pointless. The dragging performance of  joints is so slow and laggy, its useless as a professional CAD tool. Even with a good workstation its million miles behind Creo/ProE when it comes to moving parts. I hang up on it.

anirudhnehra, 2020-06-11:
The dragging performance of joints using "move" tool is not good. Is there a way to improve it or better yet animate the assembly using python?

Thanks

realthunder, 2020-06-12:
Any chance to provide a sample file showing the slowness? Maybe it is related to the Deviation/AngularDeflection [issue](https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&start=1530#p407603)? Please try the latest 0612 release image.

Also, I can't reproduce the draft snap problem. Can you please provide a sample file for this problem as well? Thanks.

anirudhnehra, 2020-06-14:
Hi,
please see the attachment. It has a video capture where I show how motion of a slider joint in two versions of freecad. The latest upstream version and your latest upstream. AS you can see , first moving in official upstream is ok but some errors crop up in 2nd try(your upstream). some errors at bottom which I try to highlight with mouse and the movement is laggy too.
I use 4 GB AMD card. Please suggest methods to improve performance and also to check whether freeeCAD is using graphics card at all.
One more issue that I try to show is that when I click OFF-GRAPHICS area, the spacemouse stops responding. Only when I click back in graphics area, space mouse works, I want it working all the time.wherever I click on screen. 
The snapping issue is solved. I accidently had snap-working-plane ON which was causing the vertices to be projected onto draft grid.
[motion_issue.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4775973/motion_issue.zip)

Thanks
Anirudh

anirudhnehra, 2020-06-14:
Hi, Again when I try to load a model created in 0.18.4, I get these warnings at bottom:
<PropShape> PropertyTopoShape.cpp(371): Recomputation required for document 'port_edilon_full_frecadasm3_big_sketch' on geo element version change in port_edilon_full_frecadasm3_big_sketch#Body.Shape: 1.14.70200.3.0 -> 1.15.70200.3.0
<App> GeoFeatureGroupExtension.cpp(568): port_edilon_full_frecadasm3_big_sketch#Extrude001.Base(links to port_edilon_full_frecadasm3_big_sketch#Sketch005) is out of scope:port_edilon_full_frecadasm3_big_sketch#Body002 vs. <global>
<App> GeoFeatureGroupExtension.cpp(568): port_edilon_full_frecadasm3_big_sketch#Compound.Links(links to port_edilon_full_frecadasm3_big_sketch#Sketch005) is out of scope:port_edilon_full_frecadasm3_big_sketch#Body002 vs. <global>

I cannot post file due to confidential design requirements. Please give a mail ID to send it

realthunder, 2020-06-15:
Email is realthunder.dev at gmail.com

The assembly file in montion_issue.zip is authored in an older version asm3. When you open this file in my latest AppImage, there will be a dialog popping up asking for recompute. You should click yes to recompute to avoid subsequent error.

> Another error:
> Texture mapping (environment) does not maintain its status after its dialog box is closed.

> to port_edilon_full_frecadasm3_big_sketch#Sketch005) is out of scope:port_edilon_full_frecadasm3_big_sketch#Body002 vs.

These types of warning is usually not fatal. But you probably should restructure you document to avoid linking an object in a different Part/Body. It may also be false warning, which I'll need your file to confirm.

> Texture mapping (environment) does not maintain its status after its dialog box is closed.
This is how it's implemented in upstream. I didn't touch that part. Please make a feature request in the forum if you really want it.

anirudhnehra, 2020-06-15:
Hi,
Its still choppy. I again share. Note that I already saved it to the newest version of your AppImage before performing the 
[motion_issue-2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4779794/motion_issue-2.zip)
drag.

realthunder, 2020-06-15:
New image 0615 released. The drag problem should be fixed. Thanks for reporting.

anirudhnehra, 2020-06-17:
Hi, I am still unable to understand linking out of scope issue. Can't I link an object from a part to another?

realthunder, 2020-06-17:
Yes, you can, with special object like the shape binder. The SubShapeBinder (the green one) has better support for that. The warning message occurs if you reference object in other container without using these special object, e.g. reference some external geometry in sketch.

anirudhnehra, 2020-06-18:
Hi,
Please have a look at the space mouse issue I mentioned above. The space navigator does not respond when I click left mouse somewhere in treeview. I need to click back into graphics area to activate it again. I am sure its an easy fix for you.

Much thanks
Anirudh

anirudhnehra, 2020-06-18:
Hi
There is still a bit of drag problem. Please see the attachment. Its boring
[motion_issue_2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4797144/motion_issue_2.zip)

anirudhnehra, 2020-06-18:
HI,
There are more issues especially related to spaceball motion. As you can see in this screen capture, the spaceball motion is choppy in this ASM3 Appimage but its smooth in the FreeCAD upstream latest.
Also you can see sketched lines (white color) quite nicely in FreeCAD upstream but they are only half visible, half masked and unclear in ASM 3 App Image.
Both use exactly the same file (the file is saved by ASM3 Applmage)

Please have a look.

Thanks
Anirudh

realthunder, 2020-06-19:
> Hi, I am still unable to understand linking out of scope issue. Can't I link an object from a part to another? Or a body using shapebinder ?

With your `big_sketch` file. One of the warning message is like this

```
port_edilon_full_frecadasm3_big_sketch#Extrude001.Base(links to port_edilon_full_frecadasm3_big_sketch#Sketch005) is out of scope:port_edilon_full_frecadasm3_big_sketch#Body002 vs. <global>
```

The warning message uses `<document>#<objectName>.<propertyName>` to indicate the source of the error. The meaning of the message above reads something like, 
`Extrude001.Base` contains a link (`Sketch005`, aka. `sketch_pattern`) which is out of scope of `Body002`, because `Sketch005` (aka `sketch_pattern`) is in Body002, while Extrude001 is not.

In upstream, the sketch_pattern remains in body. But in my branch, it is removed from body. To see why, you can test in upstream by moving `Body002`, and you'll see `sketch_pattern` moves together, while `Extrude001` stays the same. The correct way is to use a `SubLink` or `SubShapeBinder` to bind to the sketch in the body. And then extrude the link or binder.

BTW, how did you open your `big_sketch` file in upstream. Did you compile it yourself, or using the upstream AppImage? With your Deviation and AngularDeflection setting in that file, you system will freeze when opening it in upstream AppImage.

realthunder, 2020-06-19:
> Please have a look at the space mouse issue I mentioned above. The space navigator does not respond when I click left mouse somewhere in treeview. I need to click back into graphics area to activate it again. I am sure its an easy fix for you.

But I don't have a space navigator, how am I going to test it?

realthunder, 2020-06-19:
> Also you can see sketched lines (white color) quite nicely in FreeCAD upstream but they are only half visible, half masked and unclear in ASM 3 App Image.

The half visible half masked visual is normal, as the two geometries are at about the same location. You can select either geometries to make it display on top.

With upstream rendering, the object that is created earlier will be shown. So, depending on when is your `Compound` object created, it is possible to see the compound black lines instead of the sketch white lines.

anirudhnehra, 2020-06-19:
Hi, I will check with rendering issue. What about the drag issue that I posted again. The drag is still choppy.

anirudhnehra, 2020-06-23:
Hi,
I report again, this time i checked the solving time for a sample file Its just a pivot joint so just rotating. For freecad ASM 3(your applimage), the solving times are on average 0.05s while for the main FreeCAD upstream, they are below 0.02s.
This implies that your AppImage is taking more time to solve the issue.
ASM3 values:
0.030271 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.033556 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.058446 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.0998 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.059326 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.03009 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.062818 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.112052 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.056551 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.050472 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.116875 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.222563 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.052773 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.056566 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.116793 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.069276 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.05991 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.116528 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.056188 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.058638 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.110369 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.044249 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.06557 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.125736 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.049705 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.032485 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.055718 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.119236 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.062576 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.060775 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.118157 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.056667 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.057743 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.279065 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.031996 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.05326 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.054949 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.110679 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.064821 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.5172 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.080462 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.054068 <asm3.sys> sys_slvs.py(68): dof remaining: 1
1.522909 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.133386 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.113258 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.059743 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.147307 <asm3.sys> sys_slvs.py(68): dof remaining: 1
1.140781 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.059061 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.051122 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.11737 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.069996 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.055623 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.114714 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.035834 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.03563 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.037674 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.109386 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.064452 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.053748 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.116389 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.05802 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.168225 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.05037 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.049865 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.038229 <asm3.sys> sys_slvs.py(68): dof remaining: 1
[... 22 lines cut ...]
0.018441 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018355 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.011941 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018626 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018043 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.016348 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018303 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.020878 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.286627 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.019756 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.016897 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.019148 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018383 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.016231 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018249 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018565 <asm3.sys> sys_slvs.py(68): dof remaining: 1
0.018702 <asm3.sys> sys_slvs.py(68): dof remaining: 1

I even tried with selection transparency OFF in your image but same. II also turned on SMART RECOMPUTE but no difference.
I attach the sample file.
Please investigate this issue as it is a crippling issue for ASM3 image.

Regards
Anirudh
[L_pivot_asm3_sol1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/4819520/L_pivot_asm3_sol1.zip)

realthunder, 2020-06-23:
Yes, I can see similar difference on my computer as well. Still can't find out why yet. I'll keep looking.

anirudhnehra, 2020-06-23:
I have another issue regarding reference updation of external geom. Pls see the video. The Sketch_pattern is part of skeleton before being compounded with other bodies. I am unable to reference external geom from previous sketches of same body activated.

This Appimage is useless. I give up. Too big a task for single developer. I'm tired finding bugs now and then.
again unable to upload here. I send by mail.

anirudhnehra, 2020-06-23:
I am sending file through we transfer, as both of other means don't work.

anirudhnehra, 2020-06-26:
Hi,
Any updates on the sketch update issue? This works well in Upstream FreeCAD version. Just a warning is displayed that links go out of scope.

Thanks

realthunder, 2020-07-06:
The sketch problem is fixed. Please try the latest image.

anirudhnehra, 2020-07-06:
Hi,
Is there a way to automatically update appimage without downloading it
again and again?

What was the problem by the way? Can I be sure that same linking  in some
other similar circumstance will not cause problem?

Thanks
Anirudh

realthunder, 2020-07-06:
I started to generate zsync file with 16.18 release. By right, you should be able to use AppImageUpdate to download the difference. But I just tried, and it stucks. Probably something is wrong. You can try it yourself.

anirudhnehra, 2020-07-06:
Hi,
Thanks thats a work in progress.Any updates regarding slow drag performance of ASM3?

Thanks
Anirudh

realthunder, 2020-07-07:
There is a few optimization, but not specific about the asm3 drag performance. You can try and report back.

anirudhnehra, 2020-07-12:
Hi,
I have few feature requests for ASM3 workbench, 
1> Inclusion of a "regeneration value" for every distance/angle based constraint as in Pro/Engineer or Creo. This thing essentially allows to define a reference dimension values every time the document is rebuild. We may move around certain assembly bodies and those can be at weird locations. but when we rebulid the document, it should jump back to an initial configuration.(Also, I propose configurations workbench inside ASM3 , which allows different values for a constraint/ states, suppresion of parts etc.)
2> ability to create a master sketch inside ASM3 workbench and associate part bodies to it via a coordinate system(like ASM4 workbench).  Right now I think we have to convert the sketch to draft entities and its not parametric or associative.
3> DOF indicator icon for a particular part. The icon should indicate in graphics the possible DOFs available on a body. like in Siemens NX.

Thanks

realthunder, 2020-07-16:
> 1> Inclusion of a "regeneration value" for every distance/angle based constraint as in Pro/Engineer or Creo. This thing essentially allows to define a reference dimension values every time the [...]

>(Also, I propose configurations workbench inside ASM3 , which allows different values for a constraint/ states, suppresion of parts etc.)

Are these two features related? I don't quite understand the first part. For the later, i.e. configurations, you can already do this with [Configuration Table using Spreadsheet](https://forum.freecadweb.org/viewtopic.php?p=358291#p358291).

> 2> ability to create a master sketch inside ASM3 workbench and associate part bodies to it via a coordinate system(like ASM4 workbench). Right now I think we have to convert the sketch to draft entities and its not parametric or associative.

Sure, you can use sketch. Just create a circle for any reference point, and constraint the slaves to the circle. See the below screencast for an example. It uses the sketch with two circle to place the parts.

[image]

> 3> DOF indicator icon for a particular part. The icon should indicate in graphics the possible DOFs available on a body. like in Siemens NX.

Yes, I'll work on this for sure.

anirudhnehra, 2020-07-17:
Hi,
I found this on PTC website
http://support.ptc.com/help/creo/creo_pma/usascii/index.html#page/simulate%2Fmech_des%2Fmotionsettings%2Fabout_regenvalue_tab.html%23
Also though I dont have a specific example to show, pls watch this video for this idea and furthur clarification. I am sure sowehere at the end he uses regenration value parameter. Please watch the complete video as it covers a lot of ideas related to top down design implemented in Creo Parametric that I would like to see in freecad one day.
https://www.youtube.com/watch?v=rxHTBj7Ukd8