realthunder/FreeCAD_assembly3 #300: Fixing broken Subshape Binder is not possible

ceremcem, 2020-06-16:
# Priority 

Blocker

# Description 

Selecting on the `Binder` object and its `Part` and then clicking the "Subshape Binder" button does not fix the binder, it creates a new Bound object instead. 

I guess, related to that, any new `Dimension` object is created in the root of the tree, not under the TechDraw object:

[image]

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23658 (Git)
Build type: Release
Branch: LinkDev
Hash: e320e2176549a14d074cf1031115cf71f22fa322
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

Link is sent via FC forum DM.

ceremcem, 2020-06-16:
# Reproduction

```python
Gui.Selection.clearSelection()
Gui.Selection.addSelection('driver_base','Assembly','Parts.Group004.Link010.Parts.Box001.')
Gui.Selection.addSelection('driver_base', 'Binder008')
```
Then click the "Subshape Binder".

realthunder, 2020-06-17:
> Selecting on the `Binder` object and its `Part` and then clicking the "Subshape Binder" button does not fix the binder, it creates a new Bound object instead.

I have remove Binder editing function in that button. Because it creates ambiguity when the user actually want to create a new binder to the selected binder. You can use drag and drop to add more bindings. To reset the binder, hold CTRL key before dropping, which will remove all existing binding before binds to the dropped object.

I have also removed the `ReadOnly` status in the binder `Support` property. So you can alternatively edit the property in the property editor.

> I guess, related to that, any new `Dimension` object is created in the root of the tree, not under the TechDraw object:

It is not related. But it is indeed a bug, and it's been fixed.

ceremcem, 2020-06-18:
> To reset the binder, hold CTRL key before dropping, which will remove all existing binding before binds to the dropped object.

Works nicely. 

> I have also removed the ReadOnly status in the binder Support property. So you can alternatively edit the property in the property editor.

That's also very nice to have.

> I have remove Binder editing function in that button. Because it creates ambiguity when the user actually want to create a new binder to the selected binder.

I think there should be a "Changelog-diff" like message box or something for those kinds of breaking UX changes which will be displayed on the first run after an update.

closed 2020-06-18 by ceremcem